boatbod/op25 #4: Feature: Radio ID (RID) Tag Mapping

s3m1s0n1c, 2019-01-25:
Would be cool to be able to have Tag Mapping for RID's for P25
Show RID TAGS on program and make available for Icecast Stream as well.

eg..  RID.tsv would contain the following.

RID   User
12345 678     Car 221

Instead of the RID showing in the program or icecast it would show "Car 221"

Thanks
Sonic

boatbod, 2019-02-22:
This is definitely a popular request and is on my long term todo list.  The biggest issue would be the old curses terminal which doesn't have the real-estate available to accept a RID tag.  Would be more feasible on the http terminal.

scresante, 2021-03-20:
I believe this is implemented via the tgid_tags_file column in the trunk configuration file now?

boatbod, 2021-03-20:
> I believe this is implemented via the tgid_tags_file column in the trunk configuration file now?

No the boatbod version does not handle RID mapping at this time.

scresante, 2021-03-30:
Ah. I confused Talkgroup ID with Radio ID. 

On that note, when using a whitelist, the other active TGID's show up in the System Frequency table on the web interface.
How hard would it be to map the Active TGID's to their labels as they show up? (It would be nice to know who else is chatting, that I have excluded via whitelist).

boatbod, 2021-03-30:
It wouldn't be "hard" per-se, but the issue is space and formatting.  The tags can be arbitrary length and really don't lend themselves to being inserted into the small (5 character) table used to show current activity.  For it to work reasonably well you'd need either a separately defined short tag field or to truncate the first 5 characters of the current tag.

boatbod, 2021-06-07:
RID tagging has been added for the multi_rx.py variant of P25.
See p25_rtl_example.json for the configuration parameter "rid_tags_file" which points to a file of the same format as the trunk-tags.tsv file.  i.e. two column <numeric rid><tab><radio id text tag>

closed 2021-06-07 by boatbod