boatbod/op25 #5: Feature: Lock down web gui

s3m1s0n1c, 2019-01-28:
Would be nice to be able to lock down web gui with a username and password to stop unauthorized people accessing web gui.

Thanks

khanfar, 2019-02-21:
How to acsess web gui for op25 ? And whats can i see there ?

khanfar, 2019-02-28:
Is there way to make audio come through web GUI ???

s3m1s0n1c, 2019-03-01:
> How to acsess web gui for op25 ? And whats can i see there ?

add ``-l http:0.0.0.0:8080`` to your syntax for rx.py
eg ``./rx.py --args "rtl=2" -S 2400000 -v 1 -2 -T cc.tsv -V -n -U -M meta.json -l http:0.0.0.0:8080 -N 'lna:29' ``

> Is there way to make audio come through web GUI ???
No there is no audio passed thru the web gui.. You would have to use ya speaker or stream it to an icecast server.

khanfar, 2019-03-01:
Thanks for info , but how to record log of active frequency and were i can
find it ?

hayden-t, 2020-11-23:
this would be easy, i can look into it

boatbod, 2020-12-08:
Current gui has ZERO security.  Probably needs rewriting from scratch to make this feasible.

closed 2020-12-08 by boatbod