boatbod/op25 #8: Request: Icecast2 integration - example files

dthomson134, 2019-02-04:
Can you please add an example to feed audio to Icecast2.  
From my reading of the docs they indicate I need to alter:
meta.json
op25.liq
icecast.xml

**Great project.
Thanks for all the hard work!**

s3m1s0n1c, 2019-02-04:
All you need to do is edit meta.json & op25.liq with your icecast server details.
then run rx.py **-U** (insert other arguments here) Don't forget the -U
then run op25.liq

I find it best to run them in a screen
screen -dm -S op25 ./rx.py -U (arguments here)
screen -dm -S stream ./op25.liq

Hope that helps :)

boatbod, 2019-02-22:
Actually if you want to stream via liquidsoap you definitely **do not want** a -U option because that enables the built-in audio player.  What you need is -V -w and then run the op25.liq script in a separate terminal window.  It's all written up in the README-rpi3-liquidsoap file found in the top level op25 directory.

closed 2019-02-22 by boatbod