boatbod/op25 #10: Feature: Toggle Talk groups on and Off from Web Interface

cbellgit, 2019-02-11:
Was thinking it would be nice to have a page that could be accessed from the web interface that lists the talk groups (and descriptions) that are in the whitelist with a checkbox next to them.  Checking or unchecking the box would basically add or remove them from the whitelist.  I haven't looked at the code, so this probably isn't as easy to implement as one might hope but figured I'd throw it out there.

khanfar, 2019-02-21:
How to acsess web interface?

boatbod, 2019-02-22:
Dynamic whitelist and blacklist is something I have thought about.  I'll bear it in mind for the future, once I've got DMR support more sorted out.

khanfar, 2019-02-22:
How  to acsess web interface in op25 ?

boatbod, 2020-12-08:
Dynamic whitelist/blacklist is supported but only by entering the tgid# from the terminal (either curses or web).  GUI could use a lot of enhancing...

closed 2020-12-08 by boatbod