boatbod/op25 #11: Tuning error +-1200

khanfar, 2019-02-18:
I try to tune but it always give me thix error   help

thetoivonen, 2019-02-18:
Have you set the ppm correction for your dongle using the -q flag?

khanfar, 2019-02-18:
I set like this -q 15 -2 does this correct? -2 for phase 2 ??

thetoivonen, 2019-02-18:
Why are you using 15?

khanfar, 2019-02-18:
I test the dongle with sdr # in knowing frequency , then i correct it in ppm correction +15 , so may be this get problem with me ? I put in -q 15 , does this correct ,? What is q value for ????

boatbod, 2019-02-22:
Tuning error +/- 1200 or 2400 means you are off-frequency slightly.  Adjust ppm by +/-1 (at a time) until you get it correctly working.  You can use fine tuning too, but I recommend you keep those adjustments under +/-300.

gonzoblue, 2019-12-28:
Hopefully resurrecting this old thread is OK... is there a command-line way to get the signal centered? I can start op25 fine with -q -1 but after an hour or two i get the 1200 tuning errors. If do -q 0 or -q -2 it won't even lock on. Can you do decimal adjustments, like -1 1.2?
Or is it possible you could put a watchdog in the app that if it doesn't get any new TGIDs in X minutes it would restart the tuning? Otherwise I'll just have to restart the application regularly to make sure it is "stable" for me.

hayden-t, 2019-12-28:
Hi, yes you found the fine tune. I wrote a watchdog script that restarts on this error. But then found I was using the wrong modulation type. I'll look up the details again.

hayden-t, 2019-12-28:
Try putting "-D fsk4" on the rx.py command line and see if that makes it any more stable.

hayden-t, 2019-12-28:
https://forums.radioreference.com/threads/heres-a-script-to-restart-op25-if-a-tuning-error-crash-happens.398101/

gonzoblue, 2019-12-28:
Is there a way to tell for sure what my system is using with my SDR? Will the "wrong" one work somewhat even if the system is using the other one? From looking at the FCC license for the site its emission type is 8K70D1W, which is CQPSK [from what I can tell](https://wiki.radioreference.com/index.php/Emission_Designator).

I have it on CQPSK monitoring a P25 Phase 1 simulcast system, it has been good for a few hours after fine tuning and dropping my gain by quite a bit. I'll see if its still locked on tomorrow and maybe mess with the other modulation just to see if it is better or worse.

And I saw your script, but running the first part of the command before the pipe came up with nothing for OP25, so my logging is low/off... its on a raspberry pi so most logging is off to save the SD card by default I believe.

hayden-t, 2019-12-28:
So I don't think you can tell from SDR. And suprisingly I found out both can work, but if it's wrong it will bomb out after some 'random' time. My designator is 10K1F9W. As for the logging I thought I had it on default pi settings too. But I can check it in a few days.

boatbod, 2019-12-28:
The datascope (plot #4) will have different sized eye pattern when you
are receiving fsk4 vs cqpsk.  Usually it's also a fair bet that any
system with TDMA (phase 2) channel encoding is going to be cqpsk.

While you can sometimes get away with having the wrong modulation scheme
selected while tuned to a control channel, decode breaks almost
immediately when tuning to voice traffic, so if you see lots of "voice
channel timeout" messages you may want to switch modulation and try again.

As far as tuning drifting over time, this is almost always a problem
with the dongle.  I run about 6 different sdr devices and none drift
except the really old one that is non-tcxo.  Even that is stable for
days once you get it up to temperature.

Graham

jkldgoefgkljefogeg, 2020-04-13:
How does these parameters affect tuning error?

- -x
- -X
- -q
- -d

boatbod, 2020-04-13:
-X turns on frequency tracking.   Assuming there is a control channel tuned
close enough for the demodulator to see, it will auto adjust ppm correction
and find tuning to lock the signal +/- 200hz

-q is manual ppm correction, usually a positive or negative integer.

-d is manual fine tune, also a positive or negative integer.   Should never
need to be set more than +/-350 as you'd adjust the ppm instead.

-x is audio gain adjustment,  usually a positive floating point number
where 1.0 is unity and 2.0 is x2 multiplier etc.   Too high will cause
clipping.

Graham

jkldgoefgkljefogeg, 2020-04-19:
Thanks. I meant to ask about the difference betwen `-o`  and `-d`

With rtl-sdr.com V3, `-X` by itself works out of box.  

With TXCO, `-q` is usually not needed or max +-2?

I used `-o 25000` but now control frequency is a bit off. I can't seem to figure out the right number for `-o` any more.  With web interface, I can use +/- 100 with mixer to get it working. 

```
Frequency error: 85 Hz. (approx)
Fine tune offset: -299
```
should I just use `-d -300`?

khanfar, 2020-04-29:
Is there any update in op25 to make it auto ppm tuning in future ?

boatbod, 2020-12-08:
I experimented with auto-tuning in rx.py and the results were less than stellar.

closed 2020-12-08 by boatbod