boatbod/op25 #13: Is there way to record active voice ???

khanfar, 2019-02-19:
Is there way to record active voice? and active frequency history log ?

boatbod, 2019-02-22:
tgid/freq log can be achieved by turning logging up to "-v 1" then look in the log for "voice update" entries.

Unfortunately there really isn't a way to record the voice unless you use an application such as liquidsoap to archive to file as well as sending to the audio subsystem.

closed 2019-02-22 by boatbod

khanfar, 2019-02-22:
Please is there instructions how to figure liquidsoap with op25 ?

khanfar, 2019-02-22:
You mean add -v 1 to ./rx.py --args .... command ? Were should i see the
log file location ?

khanfar, 2019-02-22:
i try to use installation liquidsoap.sh from this link : https://gist.github.com/dtcooper/a1a9d6b1b96a01191514cfbb425e5f35
but in final installation on my ubuntu 18.04 i got this error :
W: Not_found
W: Not_found
E: Cannot find external tool 'ocamlbuild'
E: Failure("1 configuration error")

please give us how use this and install with op25