boatbod/op25 #16: Audio through http page ?

khanfar, 2019-02-28:
I view from my phone and control op25 in http local ip adress with port 8080 , but how to make audio push to http or make voice come through my phone that connected to http ip ????

tarantir, 2019-03-01:
You need to use icecast in combination with http. See below for an example.

./rx.py --args rtl=0 -N lna:49 -T my.tsv -q -1 -S 2400000 -l http:0.0.0.0:8999 -w -M meta.json -n -V -v 5 2> my.log

khanfar, 2019-03-01:
Icecast need to be installed first in ubuntu then use this command ?? Please explain more details steps how to do that for beginner users , thanks in advance .

tarantir, 2019-03-02:
Follow the instructions detailed in README-rpi3-liquidsoap

hayden-t, 2020-11-23:
Im not sure hes talking about icecast, if hes controlling via localhost than he can just connect to the loop back.

closed 2020-12-09 by boatbod