boatbod/op25 #17: Is there way to auto tune and auto ppm to exct frequency ?

khanfar, 2019-03-01:
Is there way to auto tune to exactly best frequency? So no use manual tuning?? And for ppm ?

JohnDN90, 2019-07-18:
I'm interested in this as well.  It'd be particularly useful for a in-car scanner using a RaspberryPi to have OP25 autoadjusting the frequency as the RTL-SDR heats up and the frequency drifts.

EricTendian, 2020-05-01:
I think this issue can be closed now, it looks like the feature has been added (see point 3): https://github.com/boatbod/op25/blob/master/op25/gr-op25_repeater/apps/README-configuration

khanfar, 2020-05-04:
im not understand this part :     ./setTrunkFreq 773.84375,774.19375
If you need multiple frequencies because your control channel moves around,
   place them in a comma separated list as follows (with no whitespace)  

i edit trunk.tsv file and when i open it i see column table then i do this 

control channel list 850.xxx  offset 0    NAC 0xxxx   Modulation c4fm 

you mean if i have another control frequency add it behind the main frequency in same array ? like 
850.xxx , 860.xxx  
this you mean ??? what the two frequencies have different NAC !

closed 2020-12-08 by boatbod