boatbod/op25 #23: Will OP25 ever be able to stream to a shoutcast server?

Torchwoodhosting, 2019-06-11:
Does anyone know if OP25 ever be able to stream to a shoutcast server?

cwagner17, 2019-06-11:
Already possible see the below thread.

https://forums.radioreference.com/threads/setting-up-op25-to-stream-to-broadcastify.368322/

Torchwoodhosting, 2019-06-11:
Thanks for the reply but what I am looking for is a way to stream it to an external WHMSONIC soutcast server not a shoutcast server located on the same machine....I should have mentioned that in my original post and btw since I forgot to mention i'm running op25 on a ubuntu 18.04.2 LTS machine and i am in no way a linix expert, I know a little but not an expert at all (so I would say dangerous LOL) Sorry i didnt make my concept clear but that what happens when you work 3rd shift and write something before sleeping :(

boatbod, 2020-12-09:
op25 has an interface to liquidsoap (See README-rpi3-liquidsoap).  From there you can stream to anything that liquidsoap is compatible with, including shoutcast.

closed 2020-12-09 by boatbod