boatbod/op25 #24: Dongle USB minor movement/ interruption suspends program and requires restart

Nokoa, 2019-07-21:
I have used OP25 with many different RTL dongles on many different computers with different versions of Ubuntu Linux and in all of them I had the same issue. Sometimes the USB extension cables move slightly interrupting the connection of the dongle to OP25 essentially stopping OP25 so the numbers no longer increment on the terminal window.  Simply restarting OP25 would fix it, no need to touch the dongle at all. I would often not notice this right away because the stream would just be quiet with no indication other than visually looking at the terminal window of OP25.

I would super appreciate it if OP25 can attempt to reestablish connection on its own rather than me having to restart it. 

Thank you for all that you do to OP25 I love this very much :)

thetoivonen, 2019-07-21:
See my comment on issue #6 for a possible workaround.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/boatbod/op25/issues/24?email_source=notifications\u0026email_token=AEKXHU6474T6WSG26WKWKRLQAPKJVA5CNFSM4IFQMMXKYY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4HAO4M6Q",
"url": "https://github.com/boatbod/op25/issues/24?email_source=notifications\u0026email_token=AEKXHU6474T6WSG26WKWKRLQAPKJVA5CNFSM4IFQMMXKYY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4HAO4M6Q",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

boatbod, 2021-12-19:
I have a solution for this problem if you are using systemctl to start & run op25.  In the latest git commits I have added two files which can be used to set up a custom UDEV rule that will invoke a shell script when the dongle misbehaves and disconnects from the USB bus.  

Copy the file 5-sdr_hw.rules into /etc/udev/rules/d , then edit is so the script points to the op25 'apps' directory and the appropriate product number is defined for your SDR hardware.  Restart the machine.  The example includes an Airspy and an RTL, but your product numbers may be different.  Look at "lsusb" to find the product number.

closed 2021-12-19 by boatbod