boatbod/op25 #25: OP25 STOPS WITH CMAKE 2 ERRORS

simplio, 2019-08-05:
I am having the worse time trying to get OP25 to completely build. I am not sure if cmake is the issue. CMAKE v3.13.3. LUBUNTU 18.10 is the distro. I am begging for help to make this work. 

Here is the screen output

ddejr@ddejr-pc:~/op25$ ./install.sh -Wno-dev
Hit:1 http://security.ubuntu.com/ubuntu cosmic-security InRelease
Hit:2 http://us.archive.ubuntu.com/ubuntu cosmic InRelease                  
Hit:3 http://us.archive.ubuntu.com/ubuntu cosmic-updates InRelease          
Hit:4 http://us.archive.ubuntu.com/ubuntu cosmic-backports InRelease
Reading package lists... Done
Reading package lists... Done
E: You must put some 'source' URIs in your sources.list
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.5ubuntu2).
cmake is already the newest version (3.12.1-1).
doxygen is already the newest version (1.8.13-10ubuntu1).
libpcap-dev is already the newest version (1.8.1-6ubuntu1).
pkg-config is already the newest version (0.29.1-0ubuntu2).
python-numpy is already the newest version (1:1.14.5-1ubuntu4).
python-waitress is already the newest version (1.1.0-1).
gnuradio is already the newest version (3.7.13.4-1build1).
gnuradio-dev is already the newest version (3.7.13.4-1build1).
gr-osmosdr is already the newest version (0.1.4-14build5).
libhackrf-dev is already the newest version (2018.01.1-2).
libitpp-dev is already the newest version (4.3.1-9build1).
librtlsdr-dev is already the newest version (0.5.4-1).
libuhd-dev is already the newest version (3.12.0.0-3build1).
swig is already the newest version (3.0.12-1.2).
git is already the newest version (1:2.19.1-1ubuntu1.1).
python-requests is already the newest version (2.18.4-2ubuntu0.18.10.1).
0 upgraded, 0 newly installed, 0 to remove and 387 not upgraded.
-- The CXX compiler identification is GNU 8.3.0
-- The C compiler identification is GNU 8.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   filesystem
--   system
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'gnuradio-runtime'
--   Found gnuradio-runtime, version 3.7.13
-- Found GNURADIO_RUNTIME: /usr/lib/i386-linux-gnu/libgnuradio-runtime.so  
-- Checking for module 'cppunit'
--   Found cppunit, version 1.14.0
-- Found CPPUNIT: /usr/lib/i386-linux-gnu/libcppunit.so;dl  
-- Found SWIG: /usr/bin/swig3.0 (found version "3.0.12") 
-- Found PythonLibs: /usr/lib/i386-linux-gnu/libpython2.7.so (found suitable version "2.7.16", minimum required is "2") 
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.16", minimum required is "2") 
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.13") found components:  doxygen dot 
CMake Warning (dev) at /usr/local/lib/python2.7/dist-packages/cmake/data/share/cmake-3.13/Modules/UseSWIG.cmake:564 (message):
  Policy CMP0078 is not set.  Run "cmake --help-policy CMP0078" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  cmake/Modules/GrSwig.cmake:159 (SWIG_ADD_LIBRARY)
  op25/gr-op25/swig/CMakeLists.txt:42 (GR_SWIG_MAKE)
[... 294 lines cut ...]
[ 89%] Building CXX object op25/gr-op25_repeater/lib/CMakeFiles/test-op25_repeater.dir/qa_op25_repeater.cc.o                                              
[ 90%] Linking CXX executable test-op25_repeater
[ 90%] Built target test-op25_repeater
Scanning dependencies of target pygen_op25_gr_op25_repeater_swig_a9103
make[2]: *** No rule to make target 'op25/gr-op25_repeater/swig/op25_repeater_swig.py', needed by 'op25/gr-op25_repeater/swig/op25_repeater_swig.pyc'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:954: op25/gr-op25_repeater/swig/CMakeFiles/pygen_op25_gr_op25_repeater_swig_a9103.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
[ 21%] Built target gnuradio-op25
[ 23%] Built target test-op25
[ 24%] Built target _op25_swig_doc_tag
[ 26%] Built target _op25_swig_swig_tag
[ 29%] Built target op25_swig_swig_compilation
[ 31%] Built target _op25_swig
[ 33%] Built target pygen_op25_gr_op25_swig_ac751
[ 34%] Built target pygen_op25_gr_op25_python_b78db
[ 34%] Built target pygen_op25_gr_op25_apps_383eb
[ 51%] Built target imbe_vocoder
[ 78%] Built target gnuradio-op25_repeater
[ 85%] Built target op25-d2460
[ 86%] Built target scan4sync
[ 87%] Built target BER
[ 90%] Built target test-op25_repeater
make[2]: *** No rule to make target 'op25/gr-op25_repeater/swig/op25_repeater_swig.py', needed by 'op25/gr-op25_repeater/swig/op25_repeater_swig.pyc'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:954: op25/gr-op25_repeater/swig/CMakeFiles/pygen_op25_gr_op25_repeater_swig_a9103.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

simplio, 2019-08-05:
Same make error trying to build op25 with cmake same result.

ddejr@ddejr-pc:~$ git clone git://github.com/boatbod/op25
Cloning into 'op25'...
remote: Enumerating objects: 231, done.
remote: Counting objects: 100% (231/231), done.
remote: Compressing objects: 100% (154/154), done.
remote: Total 4574 (delta 131), reused 155 (delta 74), pack-reused 4343
Receiving objects: 100% (4574/4574), 1.29 MiB | 1.67 MiB/s, done.
Resolving deltas: 100% (3002/3002), done.
ddejr@ddejr-pc:~$ cd op25
ddejr@ddejr-pc:~/op25$ mkdir build
ddejr@ddejr-pc:~/op25$ cd build
ddejr@ddejr-pc:~/op25/build$ cmake ../
-- The CXX compiler identification is GNU 8.3.0
-- The C compiler identification is GNU 8.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   filesystem
--   system
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'gnuradio-runtime'
--   Found gnuradio-runtime, version 3.7.13
-- Found GNURADIO_RUNTIME: /usr/lib/i386-linux-gnu/libgnuradio-runtime.so  
-- Checking for module 'cppunit'
--   Found cppunit, version 1.14.0
-- Found CPPUNIT: /usr/lib/i386-linux-gnu/libcppunit.so;dl  
-- Found SWIG: /usr/bin/swig3.0 (found version "3.0.12") 
-- Found PythonLibs: /usr/lib/i386-linux-gnu/libpython2.7.so (found suitable version "2.7.16", minimum required is "2") 
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.16", minimum required is "2") 
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.13") found components:  doxygen dot 
CMake Warning (dev) at /usr/local/lib/python2.7/dist-packages/cmake/data/share/cmake-3.13/Modules/UseSWIG.cmake:564 (message):
  Policy CMP0078 is not set.  Run "cmake --help-policy CMP0078" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  cmake/Modules/GrSwig.cmake:159 (SWIG_ADD_LIBRARY)
  op25/gr-op25/swig/CMakeLists.txt:42 (GR_SWIG_MAKE)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Build type not specified: defaulting to release.
-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   filesystem
--   system
-- AMBE Tone Synthesis Enabled
CMake Warning (dev) at /usr/local/lib/python2.7/dist-packages/cmake/data/share/cmake-3.13/Modules/UseSWIG.cmake:564 (message):
  Policy CMP0078 is not set.  Run "cmake --help-policy CMP0078" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  cmake/Modules/GrSwig.cmake:159 (SWIG_ADD_LIBRARY)
  op25/gr-op25_repeater/swig/CMakeLists.txt:42 (GR_SWIG_MAKE)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
-- Generating done
[... 278 lines cut ...]
[ 90%] Built target test-op25_repeater
Scanning dependencies of target pygen_op25_gr_op25_repeater_swig_a9103
make[2]: *** No rule to make target 'op25/gr-op25_repeater/swig/op25_repeater_swig.py', needed by 'op25/gr-op25_repeater/swig/op25_repeater_swig.pyc'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:954: op25/gr-op25_repeater/swig/CMakeFiles/pygen_op25_gr_op25_repeater_swig_a9103.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
ddejr@ddejr-pc:~/op25/build$ make
[ 21%] Built target gnuradio-op25
[ 23%] Built target test-op25
[ 24%] Built target _op25_swig_doc_tag
[ 26%] Built target _op25_swig_swig_tag
[ 29%] Built target op25_swig_swig_compilation
[ 31%] Built target _op25_swig
[ 33%] Built target pygen_op25_gr_op25_swig_ac751
[ 34%] Built target pygen_op25_gr_op25_python_b78db
[ 34%] Built target pygen_op25_gr_op25_apps_383eb
[ 51%] Built target imbe_vocoder
[ 78%] Built target gnuradio-op25_repeater
[ 85%] Built target op25-d2460
[ 86%] Built target scan4sync
[ 87%] Built target BER
[ 90%] Built target test-op25_repeater
make[2]: *** No rule to make target 'op25/gr-op25_repeater/swig/op25_repeater_swig.py', needed by 'op25/gr-op25_repeater/swig/op25_repeater_swig.pyc'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:954: op25/gr-op25_repeater/swig/CMakeFiles/pygen_op25_gr_op25_repeater_swig_a9103.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
ddejr@ddejr-pc:~/op25/build$

sirrkitt, 2019-08-13:
I'm getting the same exact issue trying to compile on a raspberry pi 3. It doesn't matter if I follow the build instructions or just use the script.

thetoivonen, 2019-08-13:
Have you tried running the script with sudo?

sirrkitt, 2019-08-13:
> 
> 
> Have you tried running the script with sudo?

Tried running the script with sudo and with root and still got the same results.

I went ahead and checked out older commits and was able to get it to build. Specifically with 25c292bc1 and e5e8fe so something after those is causing the issue.

chazanov, 2019-08-17:
Same error for me (w ArchLinux):
```
[ 93%] Building CXX object op25/gr-op25_repeater/swig/CMakeFiles/_op25_repeater_swig_swig_tag.dir/_op25_repeater_swig_swig_tag.cpp.o
[ 93%] Linking CXX executable _op25_repeater_swig_swig_tag
[ 93%] Built target _op25_repeater_swig_swig_tag
Scanning dependencies of target pygen_op25_gr_op25_repeater_swig_107e3
make[2]: *** No rule to make target 'op25/gr-op25_repeater/swig/op25_repeater_swig.py', needed by 'op25/gr-op25_repeater/swig/op25_repeater_swig.pyc'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:893: op25/gr-op25_repeater/swig/CMakeFiles/pygen_op25_gr_op25_repeater_swig_107e3.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

k9wkj, 2019-08-26:
same here
`[ 93%] Built target _op25_repeater_swig_doc_tag
make[2]: *** No rule to make target 'op25/gr-op25_repeater/swig/op25_repeater_swig.py', needed by 'op25/gr-op25_repeater/swig/op25_repeater_swig.pyc'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:1028: op25/gr-op25_repeater/swig/CMakeFiles/pygen_op25_gr_op25_repeater_swig_6e23b.dir/all] Error 2
make: *** [Makefile:130: all] Error 2`
whether script or by hand
I tried bumping back to the commit levels [sirrkitt](https://github.com/boatbod/op25/issues/25#issuecomment-520717450) had success with
but no profit or joy

this on a freshly upgraded to Buster
 _Linux desk 4.19.0-5-amd64 #1 SMP Debian 4.19.37-5+deb10u2 (2019-08-08) x86_64 GNU/Linux_

simplio, 2019-08-26:
I ended up doing a full removal of CMAKE 3.13.3 then installing CMAKE 3.10.3 from source. worked.

mathstuf, 2019-12-04:
I do see changes in the SWIG modules between 3.12.4 and 3.13.0. I'll try bisecting it here once I've gotten a working 3.12-based build and reproduced the 3.13 error.

mathstuf, 2019-12-04:
Notes:

  - dependencies not checked for by CMake:
    - pcap
    - itpp
  - `find_package(REQUIRED)` is preferred to `find_package() / if (NOT FOUND) / message(ERROR)` because `find_package(REQUIRED)` generally gives better error messages.
  - I was able to reproduce the error on 3.12.4 by doing a parallel build
  - Actually, the serial build has the same problem
  - Tentative conclusion: not a bug that 3.13 introduced (but it may have exposed it)

Running with Ninja as the generator shows this warning:

```
CMake Warning (dev):
  Policy CMP0058 is not set: Ninja requires custom command byproducts to be
  explicit.  Run "cmake --help-policy CMP0058" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  This project specifies custom command DEPENDS on files in the build tree
  that are not specified as the OUTPUT or BYPRODUCTS of any
  add_custom_command or add_custom_target:

   op25/gr-op25/swig/op25_swig_doc.i
   op25/gr-op25_repeater/swig/op25_repeater_swig_doc.i

  For compatibility with versions of CMake that did not have the BYPRODUCTS
  option, CMake is generating phony rules for such files to convince 'ninja'
  to build.

  Project authors should add the missing BYPRODUCTS or OUTPUT options to the
  custom commands that produce these files.
This warning is for project developers.  Use -Wno-dev to suppress it.
```

Which is a problem that should be addressed, but doesn't seem directly related to the files mentioned in this issue. Other than that, the build with ninja completes. So not the dependency issue I thought it might be.

Looking at the difference, it seems to be a bug in `UseSwig.cmake`. Specifically, this logic:

```cmake
  if(CMAKE_GENERATOR MATCHES "Make")
    __swig_compute_timestamp(${name} ${SWIG_MODULE_${name}_LANGUAGE}
      "${infile}" "${workingdir}" swig_generated_timestamp)
    set(swig_custom_output "${swig_generated_timestamp}")
    set(swig_custom_products
      BYPRODUCTS "${swig_generated_file_fullname}" ${swig_extra_generated_files})
    set(swig_timestamp_command
      COMMAND ${CMAKE_COMMAND} -E touch "${swig_generated_timestamp}")
```

The `BYPRODUCTS` list means nothing to non-`Ninja` generators, so it is meaningless if the generator matches `Make`. I think the fix may be to remove the `IMPLICIT_DEPENDS CXX` call and just add the outputs to the list of outputs (which is what I've usually done before). Removing `BYPRODUCTS` works at first glance, but likely will run afoul of the mentioned issue https://gitlab.kitware.com/cmake/cmake/issues/16830 in practice. Removing `IMPLICIT_DEPENDS` is likely to work, but also mean that the SWIG code is not built when it needs to be.

So, the solutions seem to involve, on the CMake side of things:

  - Remove `BYPRODUCTS` from the `Make` side of things.
  - Remove `IMPLICIT_DEPENDS` to avoid cmake#16830
  - Live with this deficiency in the Makefiles generators until it supports `add_custom_command(DEPFILE)` (and add this to `UseSwig` so that it works with the `Ninja` generator)

On the op25 side:

  - For `Make`-based generators, instead of depending on the `py` file, depend on `${swig_timestamp}` instead of the `${swig_generated_source}` Python file in `cmake/Modules/GrPython.cmake:160` (and later `add_custom_commands` in that same function).

mathstuf, 2019-12-04:
Actually, reading the comments again, it seems that 3.10 is the most recent working version. Between it and 3.11, I see this commit:

```
commit 2ee10119eac2802a234543581b1303d91b944afa
Author: Felix Schwitzer <flx107809@gmail.com>
Date:   Mon Nov 6 22:10:18 2017 +0100

    swig: fix incremental build in case of removed interface files
    
    Commit v3.8.0-rc1~123^2 (UseSWIG: Automatically scan dependencies of SWIG files
    for Makefiles, 2016-12-21) introduced automatic dependency scanning for files
    `%include`d in a swig interface definition file. This works fine as long as no
    such file is removed. But removing a dependent file breaks an incremental build
    and `make` complains about a missing dependency, see #16830.
    
    Integrate the approach proposed in the issue above into the SWIG-module, do the
    workaround in a conditional step as it arises only for Makefile generators. For
    other generators use the implementation before that commit.
    
    Fixes: #17433.
```

Links:

  - Issue: https://gitlab.kitware.com/cmake/cmake/issues/17433
  - MR: https://gitlab.kitware.com/cmake/cmake/merge_requests/1457

I'll file a new CMake issue to track this.

mathstuf, 2019-12-04:
Upstream issue to track this: https://gitlab.kitware.com/cmake/cmake/issues/20067

mathstuf, 2020-02-15:
#34 should fix this on the OP25 side (CMake still has an issue, but this workaround exists).

aschuhardt, 2020-10-06:
I'm still running into this issue...  I tried downgrading to `cmake` 3.10, and didn't have any luck.

```(master)⚡ % ./install.sh                                                                        ~/op25
warning: gnuradio-3.8.1.0-3 is up to date -- reinstalling
error: target not found: gnuradio-dev
error: target not found: gr-osmosdr
error: target not found: librtlsdr-dev
error: target not found: libuhd-dev
error: target not found: libhackrf-dev
error: target not found: libitpp-dev
error: target not found: libpcap-dev
warning: cmake-3.18.3-1 is up to date -- reinstalling
warning: git-2.28.0-1 is up to date -- reinstalling
warning: swig-4.0.2-1 is up to date -- reinstalling
error: target not found: build-essential
warning: pkgconf-1.7.3-1 is up to date -- reinstalling
warning: doxygen-1.8.20-1 is up to date -- reinstalling
warning: python-numpy-1.19.2-1 is up to date -- reinstalling
warning: python-requests-2.24.0-1 is up to date -- reinstalling
error: target not found: gnuplot-x11
-- The CXX compiler identification is Clang 10.0.1
-- The C compiler identification is Clang 10.0.1
-- Check for working CXX compiler: /usr/bin/clang++
-- Check for working CXX compiler: /usr/bin/clang++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/clang
-- Check for working C compiler: /usr/bin/clang -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Boost 1.35 found.
-- Found Boost components:
   filesystem;system
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.7.3")
-- Checking for module 'gnuradio-runtime'
--   Found gnuradio-runtime, version 3.8.1
-- Found GNURADIO_RUNTIME: /lib/libgnuradio-runtime.so
-- Checking for module 'cppunit'
--   Found cppunit, version 1.15.1
-- Found CPPUNIT: /usr/lib/libcppunit.so;dl
-- Found LOG4CPP: /usr/lib/liblog4cpp.so
-- Checking for module 'gmp'
--   Found gmp, version 6.2.0
-- Found GMP: /usr/lib/libgmpxx.so
-- Checking for module 'mpir >= 3.0'
--   Package 'mpir', required by 'virtual:world', not found
-- Found MPIR: /usr/lib/libmpirxx.so
-- Found MPLIB: /usr/lib/libgmpxx.so
-- Found VOLK: Volk::volk
CMake Warning (dev) at /usr/lib64/cmake/gnuradio/GnuradioConfig.cmake:48 (if):
  if given arguments:

    "ON"

  An argument named "ON" appears in a conditional statement.  Policy CMP0012
  is not set: if() recognizes numbers and boolean constants.  Run "cmake
  --help-policy CMP0012" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.
Call Stack (most recent call first):
  op25/gr-op25/CMakeLists.txt:97 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in /usr/lib64/cmake/gnuradio/GnuradioConfig.cmake:
  Policy CMP0011 is not set: Included scripts do automatic cmake_policy PUSH
  and POP.  Run "cmake --help-policy CMP0011" for policy details.  Use the
[... 419 lines cut ...]
-- Up-to-date: /usr/local/include/op25_repeater/dstar_tx_sb.h
-- Up-to-date: /usr/local/include/op25_repeater/fsk4_slicer_fb.h
-- Installing: /usr/local/lib/libgnuradio-op25_repeater.so
-- Installing: /usr/local/bin/scan4sync
-- Installing: /usr/local/bin/BER
-- Installing: /usr/local/lib/python3.8/site-packages/op25_repeater/_op25_repeater_swig.so
-- Set runtime path of "/usr/local/lib/python3.8/site-packages/op25_repeater/_op25_repeater_swig.so" to ""
-- Installing: /usr/local/lib/python3.8/site-packages/op25_repeater/op25_repeater_swig.py
-- Installing: /usr/local/lib/python3.8/site-packages/op25_repeater/op25_repeater_swig.pyc
-- Installing: /usr/local/lib/python3.8/site-packages/op25_repeater/op25_repeater_swig.pyo
-- Up-to-date: /usr/local/include/op25_repeater/op25_repeater/swig/op25_repeater_swig.i
-- Installing: /usr/local/include/op25_repeater/op25_repeater/swig/op25_repeater_swig_doc.i
-- Up-to-date: /usr/local/lib/python3.8/site-packages/op25_repeater/__init__.py
-- Installing: /usr/local/lib/python3.8/site-packages/op25_repeater/__init__.pyc
-- Installing: /usr/local/lib/python3.8/site-packages/op25_repeater/__init__.pyo
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/op25_repeater_vocoder.xml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/op25_repeater_gardner_costas_cc.xml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/op25_repeater_p25_frame_assembler.xml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/op25_repeater_fsk4_slicer_fb.xml
(master)⚡ % cd op25/gr-op25_repeater/apps/                                                      ~/op25
(master)⚡ % ./rx.py                                                  ~/op25/op25/gr-op25_repeater/apps
Traceback (most recent call last):
  File "./rx.py", line 54, in <module>
    import op25
ModuleNotFoundError: No module named 'op25'```

boatbod, 2020-12-08:
Seems to work under Python3 and Ubuntu 20.04 using cmake version 3.16.3
Happy to push a fix for earlier Python2/CMake versions if someone can make a pull request.

PR #34 (2020-03-01): cmake: work around CMake UseSwig behavior change
CMake 3.11 included a change which made the dependency directly on the
files not work in Makefile-based generators. Instead, depend on the
custom target provided in that case.

Fixes: #25
commit ce015555 (2020-03-01): cmake: work around CMake UseSwig behavior change

CMake 3.11 included a change which made the dependency directly on the
files not work in Makefile-based generators. Instead, depend on the
custom target provided in that case.

Fixes: #25
closed 2020-12-08 by boatbod