boatbod/op25 #28: Poor Audio Quality

Nokoa, 2019-10-02:
Thank you for all the continuous development on this project, it has been a life saver. Been using it for a few years now.

I have some public feeds of p25 phase 2 tdma and wanted to try and improve their audio quality.

I am not sure exactly how to describe this. When I was using the "heavy" original OP25 version I seemed to have some good, easy to understand audio.

But now it's a bit different.

Right now I run 2 feeds off of a laptop they play into 2 separate simulated loop back devices. It was originally too low volume so through ./audio.py I set audio gain. This is where things I think got  a little funky. I set it to 3 which is pretty high, but other wise can't hear a thing. It obviously clips sometimes. But not only that, when I look at recordings of it in audacity vs, a physical scanner recording files the scanner recordings look like they are quieter visually vs the OP25 ones which are very thick and tall. But I can hear them almost the same in volume, the OP25 ones seem to be maybe very bassy? I am not sure at all how to describe this, but if you have experienced you might understand.

In addition there would sometime be some loud chip noise at some transmissions.

Lastly sometimes a part of the transmission would get cut off, like there would be a transmission and a word in there would just go blank. Sometimes I can't hear short transmissions at all. This used to work great before, not sure what's happening now.

P25 Phase 2 is hard enough to understand when it's working perfectly, I have been comparing recordings and I saw that I have a hard time understanding what they are saying in some transmissions on OP25 but was able to understand just fine the same transmission on a scanner.

I love your OP25, and would love if you can look into some of these issues that I am having.
I might just be dumb and have messed something up, I don't know.

Thanks again for everything, I saw you started implementing patching, I really appreciate it.

unawakened, 2019-11-20:
I have similar issues. It seems to improve marginally when I run OP25 on faster hardware.

boatbod, 2020-03-01:
Nothing has changed in the codec for a long time so I doubt it's an op25 issue per-se.  Usually glitchy audio can be attributed to either running on low-spec hardware (e.g. RPi3) with excessively large sdr sample rate (-S parameter) or sometimes the Linux sound system ALSA vs Pulse audio.
The gnuplots also consume lots of cpu, so suggest making sure you are not running a plot unless you really need to see it.

closed 2020-03-01 by boatbod