boatbod/op25 #29: python3

Urganmcnurgan, 2019-11-18:
now that gnuradio is at version 3.8 which uses python 3. will op25 be getting updated for python 3? FYI: python 2.7 is considered deprecated after jan 1,2020.

jkldgoefgkljefogeg, 2020-04-13:
op25 install.sh no longer works with gnuradio 3.8 which uses Python3
https://github.com/osmocom/op25/blob/master/README-gr3.8.patch

AndrewBarfield, 2020-04-25:
Yesterday, I upgraded to Ubuntu 20.04. Running OP25 normally like this...

```
#!/bin/bash

cd ~/Projects/SDR/op25/op25/gr-op25_repeater/apps

reset

./rx.py --args 'rtl=0' \
	--crypt-behavior=2 \
	-d -101 \
	--gain-mode=1 \
	-n \
	-o 20e3 \
	-q 1 \
	-S 2560000 \
	-t \
	-T trunk.tsv \
	-U \
	-v 10 \
	-V \
	-w \
	-x 1 \
	-X \
	-2 \
	2> op25.log
```
... results in:
```
Traceback (most recent call last):
  File "./rx.py", line 49, in <module>
    from gnuradio import audio, eng_notation, gr, gru, filter, blocks, fft, analog, digital
ImportError: No module named gnuradio
```
I suspected that I needed to downgrade from GR 3.8.1.0 (Python 3.8.2) to 3.7.

After looking at this thread I copied [gr3.8.patch](https://github.com/osmocom/op25/blob/master/gr3.8.patch) to my clone of this (boatbod/op25) repository and ran `cat gr3.8.patch | patch -p1`:

```
xxx@xxxxx:~/Projects/SDR/op25$ cat gr3.8.patch | patch -p1
patching file cmake/Modules/GrSwig.cmake
patching file op25/gr-op25/CMakeLists.txt
Hunk #1 succeeded at 94 (offset 1 line).
patching file op25/gr-op25/lib/CMakeLists.txt
Hunk #1 succeeded at 56 (offset -7 lines).
patching file op25/gr-op25/swig/CMakeLists.txt
patching file op25/gr-op25_repeater/CMakeLists.txt
Hunk #1 FAILED at 92.
1 out of 1 hunk FAILED -- saving rejects to file op25/gr-op25_repeater/CMakeLists.txt.rej
patching file op25/gr-op25_repeater/lib/CMakeLists.txt
Hunk #1 succeeded at 81 (offset 14 lines).
patching file op25/gr-op25_repeater/swig/CMakeLists.txt
```

Does that patch only apply to osmocom/op25? Should I clone/patch/install it and then rebuild boatbod/op25 separately?

Any help is greatly appreciated!
Thank you in advance.

EvanKrall, 2020-06-06:
The patch above is based on a slightly different version of the code, so there are some conflicts in `op25/gr-op25_repeater/CMakeLists.txt` that need resolving -- a few lines were moved around, but that makes `patch` fail.

Here's an updated version: https://gist.github.com/EvanKrall/bd9a6088ce37b2d67b17e80716c04865

treys, 2020-06-11:
I'm trying to build op25 on a system with python3 and gnuradio 3.8.1.

When I do not apply gr3.8.patch, then op25 will build successfully, but python scripts like rx.py will not run because of the "no module named op25 found" error.

When I apply the gr3.8.patch provided by @EvanKrall, the patch is applied successfully, but I can not longer build op25. I get make errors outlined below. I have tried this patch/build against all branches, including master, python3, dev.

```
...
[ 20%] Building CXX object op25/gr-op25/lib/CMakeFiles/gnuradio-op25.dir/vc55_imbe_decoder.cc.o
[ 21%] Building CXX object op25/gr-op25/lib/CMakeFiles/gnuradio-op25.dir/value_string.cc.o
[ 22%] Building CXX object op25/gr-op25/lib/CMakeFiles/gnuradio-op25.dir/pickle.cc.o
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/liborc-0.4.so', needed by 'op25/gr-op25/lib/libgnuradio-op25.so'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:496: op25/gr-op25/lib/CMakeFiles/gnuradio-op25.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/liborc-0.4.so', needed by 'op25/gr-op25/lib/libgnuradio-op25.so'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:496: op25/gr-op25/lib/CMakeFiles/gnuradio-op25.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

Any ideas? I'm stuck for now, but will update if I find a path forward.

EvanKrall, 2020-06-11:
@treys You probably need to install liborc-0.4, e.g. `apt install liborc-0.4-0`; if it complains about header files then you'll also need `liborc-0.4-dev`

treys, 2020-06-12:
Doh! Thanks, @EvanKrall. That did the trick.

I completed the build, but I still can't get this running. It sounds like you've actually been able to get op25 running with Python3 and GNURadio 3.8? I keep getting missing module errors for op25.

`ModuleNotFoundError: No module named 'op25'`

The modules did get placed in /usr/local/lib/python3/dist-packages, but that folder is in my sys.path. I even get the ModuleNotFoundError if I try a simple 'import op25' from the python cli.

Anything else I could be missing? I'm building/running on Ubuntu 20.04.

EvanKrall, 2020-06-12:
I also wasn't able to get it running, and IIRC got the same error. Probably some code needs to change `/` to `//` to get integer division behavior.

AndrewBarfield, 2020-06-13:
Folks, I'm not able to help at the moment as I'm in process of moving. But come 21 June, I'll hop right on it (assuming the problem still exists). I really want this to work on Ubuntu 20.04.

treys, 2020-06-16:
Quick update ... some progress ... I can get rx.py to run with some minor modifications to op25_c4fm_mod.py (lines 108 and 109 integer division issues as noted by @EvanKrall); however, I get continuous "tuning error +2400" (or +1200) errors. Pretty sure I have my control frequency for op25 set correctly per waterfall data in sdrangel. Wondering if my changes messed up tuning somehow. Here are the changes to lines 108 and 109:

`impulse_response = np.fft.fftshift(np.fft.irfft(self.generator(), int(self.sample_rate)))
start = np.argmax(impulse_response) - (self.ntaps-1) // 2`

(just the int cast in 108 and the // in 109)

schizobovine, 2020-06-18:
I managed to get to a similar point as @treys but with a different error.

* I switched to the python3 branch (`git checkout python3` at the top level op25 dir)
* I applied the [python 3 patch](https://gist.github.com/EvanKrall/bd9a6088ce37b2d67b17e80716c04865) from @EvanKrall 
* Did the cmake, make, make install dance
* Modified lines 108 and 109 of `op25_c4fm_mod.py` (as @treys suggested) to force integer division.

However, I did two things different:

* Modified `rx.py` to explicitly use python3 (since the default on Ubuntu is still python2 and it's something of a PITA to change system-wide)
* Instead of altering the install path via `CMAKE_INSTALL_PREFIX=/usr`, I dropped symlinks in `/usr` to the stuff in `/usr/local` (because I really didn't want to muck around with my base system):
```
sudo ln -s /usr/local/lib/python3/dist-packages/op25 /usr/local/lib/python3/dist-packages/op25_repeater /usr/lib/python3.8/
```

This seemed to let `rx.py` get past the import errors, but I got a different error than @treys:
```
sean@nike:~/.../gr-op25_repeater/apps$ ./rx.py --args 'rtl' -N 'LNA:47' -S 2500000 -x 2 -f 857.7875e6 -o 17e3 -q -2
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
gain: name: LNA range: start 0 stop 0 step 0
setting gain LNA to 47
supported sample rates 250000-2560000 step 24000
RTL Gain of 47 set to: 48.0
Exact sample rate is: 2500000.107620 Hz
[R82XX] PLL not locked!
Traceback (most recent call last):
  File "./rx.py", line 932, in <module>
    rx = rx_main()
  File "./rx.py", line 846, in __init__
    self.tb = p25_rx_block(self.options)
  File "./rx.py", line 198, in __init__
    self.open_usrp()
  File "./rx.py", line 773, in open_usrp
    self.__set_rx_from_osmosdr()
  File "./rx.py", line 702, in __set_rx_from_osmosdr
    self.__build_graph(self.src, capture_rate)
  File "./rx.py", line 275, in __build_graph
    self.demod = p25_demodulator.p25_demod_cb( input_rate = capture_rate,
  File "/home/sean/git/op25/op25/gr-op25_repeater/apps/p25_demodulator.py", line 245, in __init__
    decimator_values = get_decim(input_rate)
  File "/home/sean/git/op25/op25/gr-op25_repeater/apps/p25_demodulator.py", line 66, in get_decim
    if q & 1:
TypeError: unsupported operand type(s) for &: 'float' and 'int'
```
There was also some error handling stuff that appeared to fail. At first I was just lacking the `python3-waitress` package, but installing that led to other errors around httplib importing. Seems separate from the above issue?
```
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/waitress/compat.py", line 119, in <module>
    import httplib
ModuleNotFoundError: No module named 'httplib'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 72, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
  File "/usr/lib/python3/dist-packages/apport/__init__.py", line 5, in <module>
    from apport.report import Report
  File "/usr/lib/python3/dist-packages/apport/report.py", line 21, in <module>
    from urllib.request import urlopen
  File "/usr/lib/python3.8/urllib/request.py", line 88, in <module>
    import http.client
  File "/home/sean/git/op25/op25/gr-op25_repeater/apps/http.py", line 31, in <module>
    from waitress.server import create_server
  File "/usr/lib/python3/dist-packages/waitress/__init__.py", line 1, in <module>
    from waitress.server import create_server
  File "/usr/lib/python3/dist-packages/waitress/server.py", line 20, in <module>
    from waitress import trigger
  File "/usr/lib/python3/dist-packages/waitress/trigger.py", line 20, in <module>
    from . import wasyncore
  File "/usr/lib/python3/dist-packages/waitress/wasyncore.py", line 54, in <module>
    from . import compat
  File "/usr/lib/python3/dist-packages/waitress/compat.py", line 121, in <module>
    from http import client as httplib
ImportError: cannot import name 'client' from partially initialized module 'http' (most likely due to a circular import) (/home/sean/git/op25/op25/gr-op25_repeater/apps/http.py)
```

table-delete, 2020-06-27:
> * Modified `rx.py` to explicitly use python3 (since the default on Ubuntu is still python2 and it's something of a PITA to change system-wide)

You can use `update-alternatives` to change the system's default quite easily. Here's the syntax I used:

`sudo update-alternatives --install /usr/bin/python python /usr/bin/python3 10`

ginomiglio, 2020-07-19:
#57 was opened to fix the swig import issue, and I have multi_rx working on Python 3. Anyone else have updates?

boatbod, 2020-12-06:
import waitress.server "httplib" error has been resolved

closed 2020-12-06 by boatbod

bendadaasamo-a11y, 2025-08-21:
> The patch above is based on a slightly different version of the code, so there are some conflicts in `op25/gr-op25_repeater/CMakeLists.txt` that need resolving -- a few lines were moved around, but that makes `patch` fail.
> 
> Here's an updated version: https://gist.github.com/EvanKrall/bd9a6088ce37b2d67b17e80716c04865
> 
> I'm still having trouble getting this to work as applied. First, I had an issue with `ModuleNotFoundError: No module named 'op25'`, which I was able to resolve by passing `-DCMAKE_INSTALL_PREFIX=/usr` to cmake (it was putting the op25 module in `/usr/local/lib/python3/dist-packages`, which isn't in my python's sys.path, but `/usr/lib/python3/dist-packages` is).
> 
> Now I'm getting `ModuleNotFoundError: No module named 'op25_swig'`.
> 
> EDIT: ok, I fixed the op25_swig error, along with a bytes/str issue in rx.py. I've updated the patch linked above.

Am sorry can you tell me how to apply the patch or I just replace it with the old one

bendadaasamo-a11y, 2025-08-21:
https://gist.github.com/EvanKrall/bd9a6088ce37b2d67b17e80716c04865 
How I can do this patch do I need just to replace it with the old one