boatbod/op25 #35: GUI in web interface - OP25

MattSmith98, 2020-03-06:
Hey Boatbod, do you plan on adding more control options in the web interface of OP25? What I would really like to have is a way in implement zones much like you would on a radio. That way when I am traveling in my truck, I can select a different zone from the web interface. For example, when I am in my college town, I can just press that button and it will scan those  towers and talkgroups. Then, when I go back home, I can press that zone button and it will scan those towers and groups.

Also maybe add a GUI for the whitelist and blacklist. For example, have buttons with the talkgroup names where you can "mute/unmute" them just by pressing a button.

Just some ideas! Wanted to hear your input! Thanks!

boatbod, 2020-04-18:
I'm not against the idea, but what we really need is a revamp of the user interface and a method to configure op25 from the gui.  It's probably not that difficult but needs someone with html5 knowledge and time to work on it.

khanfar, 2020-05-04:
boadbob its a nice idea to do that ,, hope you do it soon and find one have html5 knowledge ,, hope to make major update ,, i see many developers update there software nowadays , they investment CORONA TIME to update there staff ,, may be we all before corona we dont have time to do many staff , hope you make small time investment for this amazing op25 software , thanks you .

Dygear, 2020-05-04:
I’ve been building the HTML5 for trunk recorder. The holy grail for me would be to allow OP25 (as you are Trunk Recoders up stream) to send love as it’s benign demodulated audio to the clients in the web interface. I know enough to make that  in theory, but I don’t have a strong enough C++ back round to be helpful on the OP25 engine side. If you could add that, I could build the front end. We could make CB it just like a radio. I’d love to also be able to transmit with it, but I understand that’s probably not a helpful option for most people that don’t ha r their own radio ID to spoof / clone into a P25 system.

NebraskaCoder, 2020-11-07:
I am a ReactJS developer and can make a REALLY responsive front end. I would need to learn how to communicate with the backend, but I don't see that being much of a problem. I might fork this repo and test it on the side and see what I can come up with for the current api.

MattSmith98, 2021-01-08:
> I am a ReactJS developer and can make a REALLY responsive front end. I would need to learn how to communicate with the backend, but I don't see that being much of a problem. I might fork this repo and test it on the side and see what I can come up with for the current api.

Were you able to get something built for a interactive GUI? My goal would be to run the server on my iPad in my truck, and use it to control OP25 on the go

MattSmith98, 2021-01-10:
https://www.youtube.com/watch?v=juk9hY0pFsU&t=2s&ab_channel=SignalsEverywhere

Something like this but built into the web server would be awesome. I run everything off my ipad and the pi so I wouldnt be able to run python code on my ipad, but if that interface was built into the pi server, i could run it off of my ipad then. Maybe add zone support too? Or have a page of buttons where you can select what channels to scan? 

These are just suggestions so please dont take this as me being pushy haha

keep up the great work!

NebraskaCoder, 2021-01-29:
At this time I haven't looked into it as the back end could probably be upgraded as well to support more features. I'd be open for collaboration if boatbod (or the master branch) wanted anything official, otherwise, at some point, I'll look into a spin-off project.

khanfar, 2021-01-30:
Signal everywhere youtube channel put something interesting gui front end , check this out : 
https://youtu.be/juk9hY0pFsU
https://github.com/KR0SIV/op25mobileControlHead/

NebraskaCoder, 2021-10-05:
For this issue, see the [new-gui](https://github.com/boatbod/op25/tree/new-gui) branch. It is very much a work in progress at this time.

closed 2025-04-30 by boatbod