boatbod/op25 #37: Adding Audio on http page , and adding record activity like mp3 format ?

khanfar, 2020-04-06:
Hi , is there way to push audio in http web view ? If no hope to add this in future and is there easy way to record voice activity like mp3 or something else in one file . ?

s3m1s0n1c, 2020-05-20:
You could put a Audio Player on the page that connects to a icecast stream locally or remote..

hayden-t, 2020-11-23:
Both would be possible.

k4jk, 2026-05-02:
I am working adding audio over websocket in a branch:

https://github.com/k4jk/op25/tree/http_audio

If anyone wants to try it out/test just add a web socket destination to your channel config like this:

"destination": "ws://0.0.0.0:9000",

The logic in main.js works like this:

If the host in destination is 0.0.0.0 or 127.0.0.1, the browser substitutes window.location.hostname automatically- so it will just try to load audio from the same address in the URL. 

If the host is an explicit IP like 192.168.1.100, the browser will load audio from the specified address as-is, in case you have a reason to do it that way

Click on the Headphones Icon between the tuning and channel buttons to start the audio to your browser.

k4jk, 2026-05-10:
@boatbod 

I had to make a one-line change to main.js to guard against a null trunking key when there is a conventional-only config #283 

> I'll explore some more, but probably not until tonight.
> G
> 
> ETA:- fixed that issue (presently in dev only)

k4jk, 2026-05-10:
> Gents,
>
>
>
> I have just testing Conventional P25 with this commit and it works perfectly.   See attached p25.json used for monitoring my ZUMspot (Ham Hotspot)
>
> along with a copy of the logfile.
>
>
>
> Conventional Analog (FM) does not produce a websocket.  There appears to be a conflict with socket ports in that the underlying digital demodulator appears to get assigned ws 9000 that triggers the conflict for the analog demodulator.  Attached are my noaa_ws.json and noaa_ws.log files for review.
>
>
>
> Thank you,
>
>
>
> Bill, WA8WG

Hi Bill,

Can you open a new issue on github for the analog audio websocket issue? I think you are right, there is a conflict with the websocket so the fix will probably not be very trivial.

Also, it looks like your attachments didn't go through. You might be able to attach them to the github issue