boatbod/op25 #42: Module not found. Using Manjaro

masterplayer30, 2020-04-27:
Trying to me op25 work with Manjaro returns:

Traceback (most recent call last):
  File "./rx.py", line 54, in <module>
    import op25
ModuleNotFoundError: No module named 'op25'

Any fixes?

AndrewBarfield, 2020-04-27:
What version of Python are you using? GNURadio 3.8.1 uses Python 3 which is giving the repo problems.

AndrewBarfield, 2020-04-27:
When I run rx.py using python 3 I get something similar:

```
Traceback (most recent call last):
  File "./rx.py", line 54, in <module>
    import op25
ModuleNotFoundError: No module named 'op25'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 63, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
  File "/usr/lib/python3/dist-packages/apport/__init__.py", line 5, in <module>
    from apport.report import Report
  File "/usr/lib/python3/dist-packages/apport/report.py", line 21, in <module>
    from urllib.request import urlopen
  File "/usr/lib/python3.8/urllib/request.py", line 88, in <module>
    import http.client
  File "/home/drew/Projects/SDR/op25/op25/gr-op25_repeater/apps/http.py", line 31, in <module>
    from waitress.server import create_server
ModuleNotFoundError: No module named 'waitress'

Original exception was:
Traceback (most recent call last):
  File "./rx.py", line 54, in <module>
    import op25
ModuleNotFoundError: No module named 'op25'
```

masterplayer30, 2020-04-27:
Latest python is 3.8.2

boatbod, 2020-12-08:
Need to apply the python3 patches first.
cd ~/op25
./apply-gr3.8-patch.sh
./install.sh

closed 2020-12-08 by boatbod