boatbod/op25 #44: Signal meter implementation

JoeyAbsi, 2020-05-12:
OP25 is a great software for digital trunking!

Unfortunately, it doesn't have a signal meter (either in Db or a pictogram) to show us whether the signal is weak or strong.. This could help a lot of us and find an ideal spot to mount an antenna for the best reception.

Also, if it is possible to have it outputted through GPIO pins (Raspberry Pi) so it is possible to monitor it with LEDs.

Dygear, 2020-05-13:
How would you output it to the GPIO pins? Would you illuminated the light when you have a signal above an dB threshold? Would it just be 1 LED, or 3 LEDs making a traffic light type setup? Would it be an RGB LED? Red = Bad = Noise Floor; Yellow = Signal above Noise Floor, But with an axis deviation; Green = Strong Signal & No axis deviation?

That would also assume that this would only work on the Raspberry Pi and we would have to define out some GPIO pins for it. I think that many a separate program would be a better choice for this outright. Something really tiny that just reads an FFT over the spectrum that you're looking at. Something like that might even be able to be done in python. I know @mossman stated that he had an FFT running in 20 lines of python. Something really tiny like that would be good for something like this.

JoeyAbsi, 2020-05-13:
Well the GPIO output would be optional, but it would be more like 3 LEDs (or an RGB LED) in a traffic light light type setup. I would assume that Red = Bad reception/Very weak signal, Yellow = Weak to normal signal and Green = Very strong signal/Great reception (as if you were hundreds of meters near the signal transmission). Of course, these are only suggestions for the meter.

My main point is that a signal meter (integrated in the Web Interface) would be very useful for OP25 and its users.

I'm sure this can be done in python.

Thank you!

hayden-t, 2020-11-23:
I was literally thinking about this same topic last night as im building a portable pi based unit.

hayden-t, 2026-06-01:
what happened with this @JoeyAbsi