boatbod/op25 #46: docker installation

jkldgoefgkljefogeg, 2020-05-23:
would like to see a docker image for op25, which would solve dependency issues (python, gnuradio, rtlsdr) and make set up much easier

mosinut, 2020-05-23:
I have not had a chance yet to test it, but it looks like someone did this for the Raspberry Pi.

https://hub.docker.com/r/vassoz/sdr-balena-raspberry-pi2-op25-build

hestela, 2020-06-08:
I made a repo for the Dockerfile I'm using to run op25. https://github.com/hestela/op25-docker
`docker build -f Dockerfile-boatbod . -t op25-boatbod`
`./run-boatbod.sh`
I'm on Ubuntu 20.04 and had trouble building the osmocom and boatbod repos for op25. Since I haven' touched my tag files much I just copy them into the image but you could mount them in if you plan to edit them more.
On my rpi4, i was able to just run install.sh and everything just worked. But i think that is because it has python2 and gnuradio 3.7 by default. Starting Ubuntu 20.04, it is python 3.7 and gnuradio 3.8.

boatbod, 2020-12-08:
Due to potential for licensing issues with xMBE coded I will not be releasing any pre-built binaries of any sort.

You can build for Python3/Gnuradio-3.8 on Ubuntu 20.04 by applying the gr3.8patch before running install.sh.
cd ~/op25
./apply-gr3.8-patch.sh
./install.sh

closed 2020-12-08 by boatbod