boatbod/op25 #47: undefined reference to `imbe_vocoder::imbe_vocoder()'

blshkv, 2020-05-24:
I'm having a problem comping "python" branch it against gnuradio 3.8. Could you help please?

```
ninja -v -j4 -l0
[1/5] : && /usr/bin/x86_64-pc-linux-gnu-g++  -std=c++11  -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0    -rdynamic op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/d2460.cc.o op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/ambe.c.o op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/mbelib.c.o op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/ambe_encoder.cc.o op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/software_imbe_decoder.cc.o op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/imbe_decoder.cc.o op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/p25p2_vf.cc.o op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/rs.cc.o  -o op25/gr-op25_repeater/lib/op25-d2460  -Wl,-rpath,/var/tmp/portage/net-wireless/op25-9999/work/op25-9999_build/op25/gr-op25_repeater/lib/imbe_vocoder  op25/gr-op25_repeater/lib/imbe_vocoder/libimbe_vocoder.so && :
FAILED: op25/gr-op25_repeater/lib/op25-d2460 
: && /usr/bin/x86_64-pc-linux-gnu-g++  -std=c++11  -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0    -rdynamic op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/d2460.cc.o op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/ambe.c.o op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/mbelib.c.o op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/ambe_encoder.cc.o op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/software_imbe_decoder.cc.o op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/imbe_decoder.cc.o op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/p25p2_vf.cc.o op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/rs.cc.o  -o op25/gr-op25_repeater/lib/op25-d2460  -Wl,-rpath,/var/tmp/portage/net-wireless/op25-9999/work/op25-9999_build/op25/gr-op25_repeater/lib/imbe_vocoder  op25/gr-op25_repeater/lib/imbe_vocoder/libimbe_vocoder.so && :
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/ambe_encoder.cc.o: in function `ambe_encoder::ambe_encoder()':
ambe_encoder.cc:(.text+0x2565): undefined reference to `imbe_vocoder::imbe_vocoder()'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/ambe_encoder.cc.o: in function `imbe_vocoder::imbe_encode(short*, short*)':
ambe_encoder.cc:(.text._ZN12imbe_vocoder11imbe_encodeEPsS0_[_ZN12imbe_vocoder11imbe_encodeEPsS0_]+0x28): undefined reference to `imbe_vocoder::encode(IMBE_PARAM*, short*, short*)'
collect2: error: ld returned 1 exit status
[2/5] /usr/bin/x86_64-pc-linux-gnu-g++ -Dop25_repeater_swig_EXPORTS -I/var/tmp/portage/net-wireless/op25-9999/work/op25-9999/op25/gr-op25_repeater/lib -I/var/tmp/portage/net-wireless/op25-9999/work/op25-9999/op25/gr-op25_repeater/include -Ilib -Iinclude -Iop25/gr-op25_repeater/swig -I/var/tmp/portage/net-wireless/op25-9999/work/op25-9999/op25/gr-op25_repeater/swig -I/usr/include/gnuradio/swig -isystem /usr/include/python3.7m  -std=c++11 -fPIC   -fvisibility=hidden -MD -MT op25/gr-op25_repeater/swig/CMakeFiles/op25_repeater_swig.dir/CMakeFiles/op25_repeater_swig.dir/op25_repeater_swigPYTHON_wrap.cxx.o -MF op25/gr-op25_repeater/swig/CMakeFiles/op25_repeater_swig.dir/CMakeFiles/op25_repeater_swig.dir/op25_repeater_swigPYTHON_wrap.cxx.o.d -o op25/gr-op25_repeater/swig/CMakeFiles/op25_repeater_swig.dir/CMakeFiles/op25_repeater_swig.dir/op25_repeater_swigPYTHON_wrap.cxx.o -c op25/gr-op25_repeater/swig/CMakeFiles/op25_repeater_swig.dir/op25_repeater_swigPYTHON_wrap.cxx
ninja: build stopped: subcommand failed.
```

blshkv, 2020-05-24:
The same error using emake:

```
[ 81%] Linking CXX shared library libgnuradio-op25_repeater.so
[ 81%] Built target gnuradio-op25_repeater
[ 82%] Building CXX object op25/gr-op25_repeater/lib/CMakeFiles/test-op25_repeater.dir/test_op25_repeater.cc.o
[ 83%] Building CXX object op25/gr-op25_repeater/lib/CMakeFiles/test-op25_repeater.dir/qa_op25_repeater.cc.o
[ 84%] Linking CXX executable test-op25_repeater
[ 84%] Built target test-op25_repeater
[ 84%] Building CXX object op25/gr-op25_repeater/lib/CMakeFiles/scan4sync.dir/scan4sync.cc.o
[ 85%] Linking CXX executable scan4sync
[ 85%] Built target scan4sync
[ 86%] Building CXX object op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/d2460.cc.o
[ 87%] Building C object op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/ambe.c.o
[ 88%] Building C object op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/mbelib.c.o
[ 89%] Building CXX object op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/ambe_encoder.cc.o
[ 89%] Building CXX object op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/software_imbe_decoder.cc.o
[ 90%] Building CXX object op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/imbe_decoder.cc.o
[ 91%] Building CXX object op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/p25p2_vf.cc.o
[ 92%] Building CXX object op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/rs.cc.o
[ 93%] Linking CXX executable op25-d2460
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: CMakeFiles/op25-d2460.dir/ambe_encoder.cc.o: in function `ambe_encoder::ambe_encoder()':
ambe_encoder.cc:(.text+0x2565): undefined reference to `imbe_vocoder::imbe_vocoder()'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: CMakeFiles/op25-d2460.dir/ambe_encoder.cc.o: in function `imbe_vocoder::imbe_encode(short*, short*)':
ambe_encoder.cc:(.text._ZN12imbe_vocoder11imbe_encodeEPsS0_[_ZN12imbe_vocoder11imbe_encodeEPsS0_]+0x28): undefined reference to `imbe_vocoder::encode(IMBE_PARAM*, short*, short*)'
collect2: error: ld returned 1 exit status
make[2]: *** [op25/gr-op25_repeater/lib/CMakeFiles/op25-d2460.dir/build.make:190: op25/gr-op25_repeater/lib/op25-d2460] Error 1
make[1]: *** [CMakeFiles/Makefile
```

blshkv, 2020-05-24:
Found the problem: cmake fails to compile with 
```
-DBUILD_SHARED_LIBS=YES
```

boatbod, 2020-06-21:
I suspect a cmake dependency issue.  It's flaky and I don't know enough about cmake to be able to diagnose it and fix it.  Similar issue comes up on python2 while compiling swig using cmake newer than 3.10.3.

blshkv, 2020-06-22:
Im sure it is not cmake. One of CMakeFiles should be fixed

boatbod, 2020-06-22:
Given that it works fine on all my Ubuntu, Mint & Raspbian systems it's going to be problematic for me to reproduce without setting up a dedicated build environment to match yours.  Once you figure out the cause and proper resolution I will be happy to integrate the fix.  Thanks.

blshkv, 2020-06-22:
Are you sure you have tried with ```-DBUILD_SHARED_LIBS=YES``` in any other environment?

boatbod, 2020-06-22:
I doubt I've ever explicitly set that option, but the regular build process creates shared libs so I'm not sure what you're trying to accomplish by adding this option.

blshkv, 2020-06-22:
Shared libraries makes outcome binaries smaller. Also, you don't need to recompile your code in case if one of dependencies has been upgraded. In short, it should be a default option.
Could you please enable it and fix the makefiles?

blshkv, 2020-06-22:
You are right, it does compile share .so files by default. So this bug is more a cosmetic change, because BUILD_SHARED_LIBS is a standard variable which just happened to be enabled in my environment by default.

boatbod, 2020-06-22:
Fixed to ensure imbe_vocoder is built static.  It's a small library and doesn't need to be built shared.

closed 2020-06-22 by boatbod