boatbod/op25 #49: Python modules that are not byte-compiled

blshkv, 2020-05-24:
```
 * 
 * This package installs one or more Python modules that are not byte-compiled.
 * The following files are missing:
 * 
 *   /usr/lib/python3.7/site-packages/op25_repeater/__pycache__/__init__.cpython-37.pyc
 *   /usr/lib/python3.7/site-packages/op25_repeater/__pycache__/__init__.cpython-37.opt-1.pyc
 *   /usr/lib/python3.7/site-packages/op25_repeater/__pycache__/__init__.cpython-37.opt-2.pyc
 *   /usr/lib/python3.7/site-packages/op25_repeater/__pycache__/op25_repeater_swig.cpython-37.pyc
 *   /usr/lib/python3.7/site-packages/op25_repeater/__pycache__/op25_repeater_swig.cpython-37.opt-1.pyc
 *   /usr/lib/python3.7/site-packages/op25_repeater/__pycache__/op25_repeater_swig.cpython-37.opt-2.pyc
 *   /usr/lib/python3.7/site-packages/op25/__pycache__/__init__.cpython-37.pyc
 *   /usr/lib/python3.7/site-packages/op25/__pycache__/__init__.cpython-37.opt-1.pyc
 *   /usr/lib/python3.7/site-packages/op25/__pycache__/__init__.cpython-37.opt-2.pyc
 * 
 * Please either fix the upstream build system to byte-compile Python modules
 * correctly, or call python_optimize after installing them.  For more
 * information, see:
 * https://wiki.gentoo.org/wiki/Project:Python/Byte_compiling

```

boatbod, 2020-06-21:
I've no idea how cpython-37 gets made as it does not exist in my build tree.  Presumably it's unique to python3 which itself is largely an unknown until we get some ubuntu distros using it as standard.

Happy to look at a patch that might fix this, but as it stands there's not much I can do for you.

blshkv, 2020-06-22:
You should be able to reproduce with any python3

boatbod, 2020-06-22:
Actually it's not quite that simple.  You need a complete system built around python3 rather than python2 in order for the build tools and cmake stuff to work properly.  Just having python3 installed on (for example) ubuntu-16.04 does not help.

blshkv, 2020-06-22:
can you go to the link which I send and check how to enable it? It should be as simple as adding "-m compileall"

boatbod, 2021-03-20:
I left this open for a while in case I experienced any issues with byte compiling, but I have yet to experience any.  Python is supposed to byte-compile on an as-needed basis.

closed 2021-03-20 by boatbod