boatbod/op25 #55: NBFM Squelch

lwvmobile, 2020-07-15:
Hello,

Writing to report an issue I'm having. Currently, I have a json config file with a channel set up for monitoring an analog channel and I have not been able to set the squelch so that the channel is unmuted at the appropriate squelch threshold. It is currently either always on with static and voice, or not on at all. I have tried changing the numbers in nbfm_squelch, and also in enable_analog tried between on, auto, and even off to see if the channel would behave as it should with audio only on voice and no static. Here is my current config file that I am testing this with. using enable_analog=on results in constant static when no voice, and enable_analog=auto results in no voice, even with what I see are appropriate squelch values, and I've even dropped the squelch really far down just to see if it would trigger and it never does. Thanks for nay help.

`{
    "channels": [
        {
            "demod_type": "fsk4",
            "device": "rtl_1",
            "frequency": 154987500,
            "demod_type": "fsk4", 
            "destination": "udp://127.0.0.1:23456", 
            "excess_bw": 0.2, 
            "filter_type": "rrc", 
            "if_rate": 24000, 
            "name": "Suwannee Fire Rescue DMR", 
            "plot": "fft", 
            "symbol_rate": 4800,
            "enable_analog": "off"
        },
        {
            "demod_type": "fsk4",
            "device": "rtl_1",
            "frequency": 155070000,
            "demod_type": "fsk4", 
            "destination": "udp://127.0.0.1:23458", 
            "excess_bw": 0.2, 
            "filter_type": "widepulse", 
            "if_rate": 24000, 
            "name": "Suwannee Sheriff Analog", 
            "plot": "fft", 
            "symbol_rate": 4800,
            "enable_analog": "on",
            "nbfm_deviation": 4000,
            "nbfm_squelch": -30
        }

        
    ], 
    "devices": [
        {
            "args": "rtl=1", 
            "frequency": 155200000, 
            "gains": "LNA:26", 
            "name": "rtl_1", 
            "offset": 0, 
            "ppm": 0.5, 
            "rate": 1024000, 
            "tunable": false
        }
        
    ]
    
}`

marklt1, 2020-07-15:
The parameter in the README file is called out as "nbfm_squelch" while the code is looking for "nbfm_squelch_threshold". Changing the parameter to **nbfm_squelch_threshold** and that should resolve your issue.
I had the same issue as posted in https://forums.radioreference.com/threads/op25-feature-update.409000/#post-3334614

lwvmobile, 2020-07-15:
@marklt1 Thanks, I tried that and that did the trick. I knew I should have checked the forum, but I probably would have missed the post entirely trying to search for it. Thanks again!

closed 2020-12-08 by boatbod