boatbod/op25 #56: unexpected keyword argument 'nbfm_ctrl' - DMR Trunking from example file

lwvmobile, 2020-07-16:
Not sure what I'm doing wrong, or if there is something up in the multi_rx.py code, attempted to run a json file built from the dmr_rtl_example.json but came up with this error.

------

$ ./multi_rx.py -c 3dmr_trunking.json -v 10
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Traceback (most recent call last):
  File "./multi_rx.py", line 804, in <module>
    rx = rx_main()
  File "./multi_rx.py", line 774, in __init__
    self.tb = rx_block(options.verbosity, config = byteify(config))
  File "./multi_rx.py", line 415, in __init__
    self.configure_trunking(config['trunking'])
  File "./multi_rx.py", line 497, in configure_trunking
    self.trunk_rx = self.trunking.rx_ctl(frequency_set = self.change_freq, slot_set = self.set_slot, nbfm_ctrl = self.nbfm_control, debug = self.verbosity, chans = config['chans'])
TypeError: __init__() got an unexpected keyword argument 'nbfm_ctrl'

----

Here is the contents of my json file

----
{
    "channels": [
        {
            "device": "rtl_0",
            "frequency": 464726400,
            "demod_type": "fsk4", 
            "destination": "", 
            "excess_bw": 0.2, 
            "filter_type": "rrc", 
            "if_rate": 24000, 
            "name": "DMR Control Channel", 
            "plot": "fft", 
            "symbol_rate": 4800
        },
        {
            "device": "rtl_1",
            "demod_type": "fsk4", 
            "destination": "udp://127.0.0.1:23458", 
            "excess_bw": 0.2, 
            "filter_type": "rrc", 
            "if_rate": 24000, 
            "name": "DMR Voice Channel", 
            "plot": "fft", 
            "symbol_rate": 4800
        }
    ], 
    "devices": [
        {
            "args": "rtl=0", 
            "frequency": 464500000, 
            "gains": "LNA:49", 
            "name": "rtl_0", 
            "offset": 0, 
            "ppm": 0.5, 
            "rate": 960000, 
            "tunable": false
        },
        {
            "args": "rtl=1", 
            "frequency": 464500000, 
            "gains": "LNA:49", 
            "name": "rtl_1", 
            "offset": 0, 
            "ppm": 0.5, 
            "rate": 960000, 
            "tunable": true
        }
    ],
    "trunking": {
        "module": "tk_trbo.py",
        "chans": [
            {
                "lcn": 1,
                "frequency": 462451000,
                "cc": 5
            },
            {
                "lcn": 2,
                "frequency": 464726000,
                "cc": 5
            },
            {
                "lcn": 3,
                "frequency": 464825000,
                "cc": 5
            },
            {
                "lcn": 4,
                "frequency": 464825000,
                "cc": 5
            }
        ]
    }
}
---

I've also attempted to run the example provided, but get the same error. thanks for any help.
p.s. sorry for opening up two issues in two days time.

mike021361, 2020-08-01:
Same for me.

Leap 15.1

python -V
Python 2.7.17

Traceback (most recent call last):
  File "./multi_rx.py", line 807, in <module>
    rx = rx_main()
  File "./multi_rx.py", line 777, in __init__
    self.tb = rx_block(options.verbosity, config = byteify(config))
  File "./multi_rx.py", line 418, in __init__
    self.configure_trunking(config['trunking'])
  File "./multi_rx.py", line 500, in configure_trunking
    self.trunk_rx = self.trunking.rx_ctl(frequency_set = self.change_freq, slot_set = self.set_slot, nbfm_ctrl = self.nbfm_control, debug = self.verbosity, chans = config['chans'])
TypeError: __init__() got an unexpected keyword argument 'nbfm_ctrl'

gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib64/gcc/x86_64-suse-linux/7/lto-wrapper
OFFLOAD_TARGET_NAMES=hsa:nvptx-none
Target: x86_64-suse-linux
Configured with: ../configure --prefix=/usr --infodir=/usr/share/info --mandir=/usr/share/man --libdir=/usr/lib64 --libexecdir=/usr/lib64 --enable-languages=c,c++,objc,fortran,obj-c++,ada,go --enable-offload-targets=hsa,nvptx-none=/usr/nvptx-none, --without-cuda-driver --enable-checking=release --disable-werror --with-gxx-include-dir=/usr/include/c++/7 --enable-ssp --disable-libssp --disable-libvtv --disable-libcc1 --disable-plugin --with-bugurl=https://bugs.opensuse.org/ --with-pkgversion='SUSE Linux' --with-slibdir=/lib64 --with-system-zlib --enable-libstdcxx-allocator=new --disable-libstdcxx-pch --enable-version-specific-runtime-libs --with-gcc-major-version-only --enable-linker-build-id --enable-linux-futex --enable-gnu-indirect-function --program-suffix=-7 --without-system-libunwind --enable-multilib --with-arch-32=x86-64 --with-tune=generic --build=x86_64-suse-linux --host=x86_64-suse-linux
Thread model: posix
gcc version 7.5.0 (SUSE Linux)

RRDB:
https://www.radioreference.com/apps/db/?siteId=26840

ers-oci.rtl.cfg.json:
{
  "channels": [
    {
      "device": "rtl0",
      "demod_type": "fsk4",
      "destination": "",
      "excess_bw": 0.2,
      "filter_type": "rrc",
      "if_rate": 24000,
      "name": "ERS-OCI-Control",
      "plot": "",
      "symbol_rate": 4800
    },
    {
      "device": "rtl1",
      "demod_type": "fsk4",
      "destination": "udp://127.0.0.1:23456",
      "excess_bw": 0.2,
      "filter_type": "rrc",
      "if_rate": 24000,
      "name": "ERS-OCI-Voice",
      "plot": "",
      "symbol_rate": 4800
    }
  ],
  "devices": [
    {
      "args": "rtl=00000000",
      "frequency": 452462500,
      "gains": "LNA:42",
      "name": "rtl1",
      "offset": 0,
      "ppm": 21,
      "rate": 960000,
      "tunable": true
    },
    {
      "args": "rtl=00000012",
      "frequency": 452462500,
      "gains": "LNA:39",
      "name": "rtl0",
      "offset": 0,
      "ppm": 0,
      "rate": 960000,
      "tunable": true
    }
  ],
    "trunking": {
        "module": "tk_trbo.py",
        "chans": [
            {
                "lcn": 1,
                "frequency": 452462500,
                "cc": 0
            },
            {
                "lcn": 2,
                "frequency": 451787500,
                "cc": 0
            },
            {
                "lcn": 3,
                "frequency": 451912500,
                "cc": 0
            },
            {
                "lcn": 4,
                "frequency": 451987500,
                "cc": 0
            },
            {
                "lcn": 5,
                "frequency": 463537500,
                "cc": 0
            },
            {
                "lcn": 6,
                "frequency": 452337500,
                "cc": 0
            },
            {
                "lcn": 7,
                "frequency": 461462500,
                "cc": 0
            },
            {
                "lcn": 8,
                "frequency": 461787500,
                "cc": 0
            },
            {
                "lcn": 9,
                "frequency": 462112500,
                "cc": 0
            },
            {
                "lcn": 10,
                "frequency": 463537500,
                "cc": 0
            },
    
 
            {
                "sysname": "ERS-OCI",
                "control_channel_list": "452.4625",
                "tgid_tags_file": "ers-oci.tsv",
                "tgid_hold_time": 2.0,
                "blacklist": "",
                "whitelist": "",
                "bandplan": "400"
            }
                 ],
    "audio": {
        "module": "sockaudio.py",
        "instances": [
            {
                "instance_name": "audio0",
                "device_name": "default",
                "udp_port": 23456,
                "audio_gain": 1.0,
                "number_channels": 1
            }
        ]
    },
    "terminal": {
        "module": "terminal.py",
        "terminal_type": "curses",
        "#terminal_type": "http:127.0.0.1:8080",
        "curses_plot_interval": 0.2,
        "http_plot_interval": 1.0,
        "http_plot_directory": "../www/images"
    }
    }
}

This worked fine prior to the update from 11 days ago.

Please ask if I can provide more info.

boatbod, 2020-12-08:
Confirmed/duplicated the issue.  Fix committed to "dev" branch and will roll up to master at next release point.

closed 2020-12-08 by boatbod