boatbod/op25 #62: OP25 Refresh rate on Raspberry Pi 3

JoeyAbsi, 2020-08-09:
I am finding the 1 Hz refresh rate very slow on my raspberry pi 3 B and was wondering if there was a way to increase it as I want. 

I am only running OP25 on my Pi  and it is using 20% of the cpu. I would like to increse the refresh rate at least to 2 Hz but up to 5 Hz.

Thank you

hayden-t, 2020-11-23:
in rx.py you can set it with _def_interval = 1.0    # sec

JoeyAbsi, 2020-11-23:
Do I only have to add this line at the end of the file or are there other things that need to be done?

hayden-t, 2020-11-23:
actually that i believe is just for http interface , are you using http or terminal ?

JoeyAbsi, 2020-11-23:
I am using the http interface.

hayden-t, 2020-11-23:
ok, yeah find that line https://github.com/boatbod/op25/blob/dbadbb625a7d59b9bf9c5171c291185a841148de/op25/gr-op25_repeater/apps/rx.py#L83

and change the value. bear in mind, it writes a image each time to _def_file_dir, which will default be on your sd, and even when you you close the browser it will keep writing new images each interval unless you toggle the graph off first. this constant writing to sd will eventually use up their limited write cycles, ideally it should be on a ram drive/temp something ill look into in relation to my issue https://github.com/boatbod/op25/issues/70

JoeyAbsi, 2020-11-23:
I changed the value to 0.25 (seconds) but it didn't seem to change anything.
Is there a minimal value that can be entered?

hayden-t, 2020-11-23:
just looking into it

hayden-t, 2020-11-23:
ok, so that line we changed already affects how often the images are generated, but you still need to adjust the update rate in the javascript code for the website, to set how often it checks for latest image.
https://github.com/boatbod/op25/blob/dbadbb625a7d59b9bf9c5171c291185a841148de/op25/gr-op25_repeater/www/www-static/main.js#L515
change it to 250

JoeyAbsi, 2020-11-23:
Do I change line 83 back to 1.0 seconds?

hayden-t, 2020-11-23:
no you need both, one is the frequency images are generated, and the other is how frequent the website checks for latest image.
here is how to make images dir a ramdisk to save your sd and for better performance
https://github.com/boatbod/op25/issues/70#issuecomment-732471698https://github.com/boatbod/op25/issues/70#issuecomment-732471698

hayden-t, 2020-11-23:
ideally they should be both linked as one variable, but not yet

JoeyAbsi, 2020-11-23:
It worked. Thanks for the help!

closed 2020-11-23 by JoeyAbsi

marklt1, 2020-11-24:
> ok, yeah find that line
>
> https://github.com/boatbod/op25/blob/dbadbb625a7d59b9bf9c5171c291185a841148de/op25/gr-op25_repeater/apps/rx.py#L83
>
> and change the value. bear in mind, it writes a image [...]

To avoid excessive SD wear, I have been using log2ram for my logs and image folders  - https://github.com/azlux/log2ram
It caches reads/writes and flushes writes to the SD periodically (hourly or when shutting down).
More info in this article: https://mcuoneclipse.com/2019/04/01/log2ram-extending-sd-card-lifetime-for-raspberry-pi-lorawan-gateway/

I use the following script to install and configure on my RPi OP25 units

curl -Lo log2ram.tar.gz https://github.com/azlux/log2ram/archive/master.tar.gz
tar xf log2ram.tar.gz
cd log2ram-master
sudo chmod +x install.sh && sudo ./install.sh
sudo sed -i 's|"/var/log"|"/var/log;/home/pi/op25/op25/gr-op25_repeater/www/images"|g' /etc/log2ram.conf
sudo reboot

I send my OP25 logs to **/var/log/scanner.log**