boatbod/op25 #64: multi_rx sends audio to wireshark instead of P25 frames

jsherman256, 2020-08-24:
I'm trying to use wireshark to read the raw P25 frames from a trunked TDMA Phase-2 system. If I use `rx.py` with the control frequency passed, I am able to get valid P25 frames in wireshark, but only Trunk Control type frames.

If I use `multi_rx.py`, it seems to send decoded audio to wireshark instead of the LDU1 and LDU2 frames. Wireshark is interpretting them as p25cai, but they have nonsense in the headers (including frame synchronization field). I also notice this in the terminal output:
```op25_audio::open_socket(): enabled udp host(172.17.0.4), wireshark(23456), audio(23456)```

I am able to record the audio on this system via trunk-recorder, so I'm fairly certain it's not a reception/hardware/driver on my side

Am I missing something obvious here? Is there a way to feed all P25 frames to wireshark from a trunked system?

jsherman256, 2020-08-24:
This is the associated JSON configuration file when running `multi_rx.py`
```json
{
    "channels": [
        {
            "name": "Voice_ch1", 
            "device": "sdr0",
            "trunking_sysname": "FiRST Talbot",
            "meta_stream_name": "stream_0",
            "demod_type": "cqpsk", 
            "destination": "udp://172.17.0.4:23456", 
            "meta_stream_name": "",
            "excess_bw": 0.2, 
            "filter_type": "rc", 
            "frequency": 769081250, 
            "if_rate": 24000, 
            "plot": "",
            "symbol_rate": 4800,
            "enable_analog": "off",
            "blacklist": ""
        },
        {
            "name": "Voice_ch2", 
            "device": "sdr0",
            "trunking_sysname": "FiRST Talbot",
            "meta_stream_name": "stream_1",
            "demod_type": "cqpsk", 
            "destination": "udp://172.17.0.4:23456", 
            "meta_stream_name": "",
            "excess_bw": 0.2, 
            "filter_type": "rc", 
            "frequency": 769368750, 
            "if_rate": 24000, 
            "plot": "",
            "symbol_rate": 4800,
            "enable_analog": "off",
            "blacklist": ""
        },
        {
            "name": "Voice_ch3", 
            "device": "sdr0",
            "trunking_sysname": "FiRST Talbot",
            "meta_stream_name": "",
            "demod_type": "cqpsk", 
            "destination": "udp://172.17.0.4:23456", 
            "meta_stream_name": "",
            "excess_bw": 0.2, 
            "filter_type": "rc", 
            "frequency": 769618750, 
            "if_rate": 24000, 
            "plot": "",
            "symbol_rate": 4800,
            "enable_analog": "off",
            "blacklist": ""
        },
	{
            "name": "Voice_ch4", 
            "device": "sdr0",
            "trunking_sysname": "FiRST Talbot",
            "meta_stream_name": "",
            "demod_type": "cqpsk", 
            "destination": "udp://172.17.0.4:23456", 
            "meta_stream_name": "",
            "excess_bw": 0.2, 
            "filter_type": "rc", 
            "frequency": 770281250, 
            "if_rate": 24000, 
            "plot": "",
            "symbol_rate": 4800,
            "enable_analog": "off",
            "blacklist": ""
        },
	{
            "name": "Voice_ch5", 
            "device": "sdr0",
            "trunking_sysname": "FiRST Talbot",
            "meta_stream_name": "",
            "demod_type": "cqpsk", 
            "destination": "udp://172.17.0.4:23456", 
            "meta_stream_name": "",
            "excess_bw": 0.2, 
            "filter_type": "rc", 
            "frequency": 770556250, 
            "if_rate": 24000, 
            "plot": "",
            "symbol_rate": 4800,
            "enable_analog": "off",
            "blacklist": ""
        },
	{
            "name": "Voice_ch6", 
            "device": "sdr0",
            "trunking_sysname": "FiRST Talbot",
            "meta_stream_name": "",
            "demod_type": "cqpsk", 
            "destination": "udp://172.17.0.4:23456", 
            "meta_stream_name": "",
            "excess_bw": 0.2, 
            "filter_type": "rc", 
            "frequency": 770831250, 
            "if_rate": 24000, 
            "plot": "",
            "symbol_rate": 4800,
            "enable_analog": "off",
            "blacklist": ""
        },
	{
            "name": "Voice_ch7", 
            "device": "sdr0",
            "trunking_sysname": "FiRST Talbot",
            "meta_stream_name": "",
            "demod_type": "cqpsk", 
            "destination": "udp://172.17.0.4:23456", 
            "meta_stream_name": "",
            "excess_bw": 0.2, 
            "filter_type": "rc", 
            "frequency": 771181250, 
            "if_rate": 24000, 
            "plot": "",
            "symbol_rate": 4800,
            "enable_analog": "off",
            "blacklist": ""
        },
	{
            "name": "Voice_ch8", 
            "device": "sdr0",
            "trunking_sysname": "FiRST Talbot",
            "meta_stream_name": "",
            "demod_type": "cqpsk", 
            "destination": "udp://172.17.0.4:23456", 
            "meta_stream_name": "",
            "excess_bw": 0.2, 
            "filter_type": "rc", 
            "frequency": 771481250, 
            "if_rate": 24000, 
            "plot": "",
            "symbol_rate": 4800,
            "enable_analog": "off",
            "blacklist": ""
        }
    ], 
    "devices": [
        {
            "args": "rtl=0", 
            "frequency": 770281000, 
            "gains": "LNA:40", 
            "name": "sdr0", 
            "offset": 0, 
            "ppm": 0.0, 
            "rate": 2400000, 
            "tunable": true 
        },
	{
            "args": "rtl=1", 
            "frequency": 773143750, 
            "gains": "LNA:40", 
            "name": "sdr1", 
            "offset": 0, 
            "ppm": 0.0, 
            "rate": 2400000, 
            "tunable": false 
        }
    ],
    "trunking": {
        "module": "tk_p25.py",
        "chans": [
            {
                "nac": "0x0",
                "sysname": "FiRST Talbot",
                "control_channel_list": "774206250, 772081250, 771781250",
                "whitelist": "",
                "blacklist": "",
                "crypt_behavior": 1
            }
        ]
    },
    "terminal": {
        "module": "terminal.py",
        "terminal_type": "curses",
        "#terminal_type": "http:172.17.0.4:8080",
        "curses_plot_interval": 0.1,
        "http_plot_interval": 1.0,
        "http_plot_directory": "../www/images"
    }
}
```

boatbod, 2020-12-07:
Legacy wireshark support ends with rx.py and will not be implemented in multi_rx.py

closed 2020-12-07 by boatbod