boatbod/op25 #66: Audio

spencerhusak, 2020-11-12:
I can not seem to be able to get audio through bluetooth and laggy audio

JoeyAbsi, 2020-11-16:
On what are you running OP25 (RPI, Ubuntu...) and is your bluetooth speaker charged, connected?
If you try to play a video on YouTube do you still have this problem?

spencerhusak, 2020-11-16:
I got it to all connect and work but i want it to auto start when the
system boots up and after it connects to the bluetooth im on raspberry pi
4. Before i had it auto starting via modifying the /etc/rc.local file but
op25 would boot before it was connected to bluetooth and than was using the
default audio via 3.5mm jack not bluetooth any suggestions

JoeyAbsi, 2020-11-16:
OK so if I understood what you wrote, you are auto-starting OP25 on a RPI4 and want it to transmit its audio to a bluetooth speaker.
If you try to listen to the audio through the audio jack, is the audio good? 
If so, is the audio good on the bluetooth speaker while playing a YouTube video for example? 
Have you tried using another speaker to check if it works any better?

spencerhusak, 2020-11-16:
My question is do i run op25 via op25.sh or manual mode?

JoeyAbsi, 2020-11-17:
They both work great but it's way easier to run it via the OP25.sh file.

spencerhusak, 2020-11-17:
Now its running perfect via the op25.sh file i had no clue about that took
a little bit of digging

spencerhusak, 2020-11-17:
So the question is how to i get the op25.sh file to boot at start up

JoeyAbsi, 2020-11-17:
I created a service:
The file is: /etc/systemd/system/op25-stream0.service (you can name it whatever you want though). 

Write this code in the service file:
[Unit]
Description=op25-ncshp
After=syslog.target network.target nss-lookup.target network-online.target
Requires=network-online.target

[Service]
User=pi
Group=pi
WorkingDirectory=/home/pi/op25/       //Make sure to select the right directory!
ExecStart=/bin/bash -- op25.sh
RestartSec=5
Restart=on-failure

[Install]
WantedBy=multi-user.target

Save this file and write the following code in the terminal:
sudo systemctl daemon-reload
sudo systemctl start OP25.service    //Or whatever your service name is...
sudo systemctl status OP25.service

This worked for me and for others so it should be no problem.

hayden-t, 2020-11-29:
this just looks like the existing recommended way to run as service :)
https://github.com/boatbod/op25/blob/master/op25/gr-op25_repeater/apps/op25-rx.service

JoeyAbsi, 2020-12-02:
It is!

closed 2020-12-06 by boatbod