boatbod/op25 #70: html interface, constant image writes will wear out sd card.

hayden-t, 2020-11-21:
So looking at how the html interface image plots works js wise, in relation to my other issue. And i noticed the images continue be generated in ./www/images regardless of if you are connected to the website or not. over time this will wear out the sd card and kill it. images should be generated on demand, i will look into it further.

hayden-t, 2020-11-21:
Once rebooted it does not generate, only clicking the plot tab starts generation, just need to work out best way to stop it

hayden-t, 2020-11-23:
maybe a work around for now is to make /www/images a tmpfs ram drive mount

putting this in /etc/fstab and then issuing mount -a will turn images folder into 10Mb ram disk

tmpfs /home/pi/op25/op25/gr-op25_repeater/www/images tmpfs nodev,nosuid,size=10M 0 0

hayden-t, 2020-11-24:
another option would be to log time of update ajax requests, and set a timeout to stop saving plots. also a function to stop all plots would help, currently there is functions to stop each one, but you need to know which is running.

marklt1, 2020-11-24:
To avoid excessive SD wear, I have been using log2ram for my logs and image folders - https://github.com/azlux/log2ram
It caches reads/writes and flushes writes to the SD periodically (hourly or when shutting down).
More info in this article: https://mcuoneclipse.com/2019/04/01/log2ram-extending-sd-card-lifetime-for-raspberry-pi-lorawan-gateway/

I use the following script to install and configure on my RPi OP25 units

curl -Lo log2ram.tar.gz https://github.com/azlux/log2ram/archive/master.tar.gz
tar xf log2ram.tar.gz
cd log2ram-master
sudo chmod +x install.sh && sudo ./install.sh
sudo sed -i 's|"/var/log"|"/var/log;/home/pi/op25/op25/gr-op25_repeater/www/images"|g' /etc/log2ram.conf
sudo reboot

I send my OP25 logs to /var/log/scanner.log

boatbod, 2020-12-13:
multi_rx.py now monitors 'update' requests from terminal and will shut down open plots when the terminal times out.

rx.py cannot have the same code readily retrofitted so feature will not be back-ported.  Suggest updating your configuration to use multi_rx as it is more flexible and allows multiple plots to be active at the same time.

Note: enhancement has been made to 'dev' branch and will be propagated to the main line in due course.

closed 2020-12-13 by boatbod