boatbod/op25 #77: Does NAC Filter Work ?

trengking, 2020-11-27:


trengking, 2020-11-27:
I have this case : 

I set System "Alpha" with freq "855.000" as the Control Channel, NAC is 100.
Surprisingly, freq 855.000 is also member of System "Bravo" as a Voice Channel, but the NAC is 200.
So the frequency 855.000 has the same signal strength (from different Systems) where my Rpi + OP25 located.

I have set the trunk.csv to tune to 855.000 and I set the NAC to 100.
Meanwhile I was waiting for the Talkgroups of System "Alpha" to be monitored, I keep getting Voice Decoded from System "Bravo" even System Bravo has different NAC from system "Alpha".

I have tried to Blacklist all the TGID from System "Alpha" within the trunk.csv file, but I keep getting voice which I don't expect to be monitored.

Any suggestions what should I do ?

boatbod, 2020-12-06:
In P25 Ph1 the c++ library code which handles framing and voice decode does not validate NAC - it only pass it on to the python trunking module where validation takes place prior to accepting TSBKs.  Unfortunately trunking has no way to stop audio from playing if LDU1/LDU2 are present on the currently tuned channel.

In P25 Ph2 the voice decoder requires a "descrambler" xor-mask to be set correctly before intelligible voice decoding can take place.  This xor-mask is derived from several parameters, one of which is the NAC.

trengking, 2020-12-06:
well noted, and much appreciated for responding to such unique case i've got.

if you want to close this thread / issue please feel free to do so, but if it is convenient for you and team, i will let this open. hoping for the best for future enhancement / improvement with of course, low priority 😁🙏 TIA

boatbod, 2020-12-11:
I have committed some code to the "dev" branch which optionally validates the NAC at the mac layer.  Setting a NAC=0 disables initial validation, but then whatever value op25 locks on to will be the one that sticks.  These changes will eventually be promoted to the mainline, but I'd like for you to test them for a bit first.

closed 2020-12-11 by boatbod

trengking, 2020-12-11:
very well, i'll do some testing then.
many thanks for everry effort you make (and also the team) to improve this.

this is just great! cheers 😁👍

trengking, 2020-12-11:
> I have committed some code to the "dev" branch which optionally validates the NAC at the mac layer. Setting a NAC=0 disables initial validation, but then whatever value op25 locks on to will be the [...]

this works as it is supposed to ...you are awesome @boatbod 
i've tested almost  all the possibilities  that have crossed in my mind : 

1. testing a primary (and known) CCh in my area with 3 different scenarios : [A] not setting any NAC (0x0), [B] setting the correct NAC and also [C] setting the incorrect NAC. i am very satisfied with the results, [A] decodes any NAC appear first, [B] & [C] only decodes the NAC that have been defined in trunk.tsv; where of course [B] resulting  voice decoded perfectly (the correct NAC has been set) and [C] nothing was decoded (because NAC mismatched)

2. testing an interfered CCh as the case i initially raised (when this network / system idle, it was decoding other networks' VC with different NAC) this CCh has the same frequency as the VC in the neighbour network. when i tried [A] of course the other network's VC will be decoded  (as before this current 'dev' commit), but when i tried [B] which is the correct NAC for this CCh, i hear nothing decoded from 'the neighbour network' which has different NAC, it was completely silenced.

conclusion : THIS COMMIT WORKS PERFECTLY :)

thank you once again @boatbod  hope this commit will be merged to master branch in the future.
hoping i can learn python from time to time, (i have no experience in programming yet) so that i can participate in developing this great OP25, instead of just asking for improvements.

cheers mate !

marklt1, 2020-12-16:
I ended up having the same issue without even knowing it with two competing NACs.
Once this code went in, I started seeing the NAC check fails which indicate my issue.
Anyhow, the code works great now that the offending NAC is identified and blocked. Job well done!

boatbod, 2020-12-16:
Thanks for the feedback.  The changes have been promoted to master.