boatbod/op25 #87: DEV HTTP Interface Errors

NebraskaCoder, 2020-12-08:
Started a new issue for the issues with the HTTP interface in DEV.

Using the command `/rx.py --nocrypt --args "rtl" --gains 'lna:36' -S 960000 -X -q 0 -v 1 -2 -V -U -T trunk.tsv -l "http:0.0.0.0:8080" 2> stderr.2` - when there is an active channel (possibly even when there isn't an active channel), if you click go to and enter a tgid, the web interface freezes and doesn't go to the TG. It freezes on the channel that was active when the goto button was click and none of the buttons work after that moment without restarting rx.py.

NebraskaCoder, 2020-12-08:
I'm not sure which HTTP interface button caused this to show in the logs:

```python
update_state: unknown command: b'hold'
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "./rx.py", line 931, in run
    self.callback(msg)
  File "/home/ggraves/op25/op25/gr-op25_repeater/apps/trunking.py", line 1128, in process_qmsg
    self.update_state(cmd, curr_time, int(msg.arg1()))
  File "/home/ggraves/op25/op25/gr-op25_repeater/apps/trunking.py", line 1487, in update_state
    assert 0 == 1
AssertionError
```

boatbod, 2020-12-08:
I will take a look, thanks.

boatbod, 2020-12-09:
Fixes committed for a variety of problems
- Python3 string/binary handling of GUI commands
- Bug in whitelist/blacklist processing (trunking.py)
- Bug in Skip/Hold processing (tk_p25.py)

closed 2020-12-09 by boatbod

alphafox02, 2020-12-09:
Thank you Boatmod. I’ll pull in latest. New to OP25 but I can say this gnuradio 3.8/python3 is working great (I hear audio and see talk groups) with my limited setup and local police being on p25 phase 1.