boatbod/op25 #93: cmake and make fail missing boost files

srs4511351, 2020-12-24:
I am trying to install op25 on a Raspberry Pi 4 that has gnuradio 3.8 and an SDRplay RSP1A.

cmake has several warnings such as:
  Target "gnuradio-op25_repeater" links to target "Boost::program_options"
  but the target was not found.  Perhaps a find_package() call is missing for
  an IMPORTED target, or an ALIAS target is missing?

but it completes and writes the build files.

make fails with these errors:
/usr/bin/ld: cannot find -lBoost::program_options
/usr/bin/ld: cannot find -lBoost::regex
/usr/bin/ld: cannot find -lBoost::thread

libboost-all-dev 1.67 is installed.

I was able to install and run it on another boot drive that has gnuradio 3.7.

Do you know what is wrong?

boatbod, 2020-12-25:
What version linux are you running and what is the output from the following command:
apt list | grep gnuradio

srs4511351, 2020-12-26:
This is mt Linux version information:
Operating System: Raspbian GNU/Linux 10 (buster)
Kernel: Linux 5.4.79-v7l+
Architecture: arm

$ gnuradio-config-info -v
3.8.2.0

$ apt list | grep gnuradio
gnuradio-dev/stable 3.7.13.4-4+b1 armhf
gnuradio-doc/stable 3.7.13.4-4 all
gnuradio/stable 3.7.13.4-4+b1 armhf
libgnuradio-analog3.7.13/stable 3.7.13.4-4+b1 armhf
libgnuradio-atsc3.7.13/stable 3.7.13.4-4+b1 armhf
libgnuradio-audio3.7.13/stable 3.7.13.4-4+b1 armhf
libgnuradio-blocks3.7.13/stable 3.7.13.4-4+b1 armhf
libgnuradio-channels3.7.13/stable 3.7.13.4-4+b1 armhf
libgnuradio-comedi3.7.13/stable 3.7.13.4-4+b1 armhf
libgnuradio-dab3.7.13/stable 0.3-4+b1 armhf
libgnuradio-digital3.7.13/stable 3.7.13.4-4+b1 armhf
libgnuradio-dtv3.7.13/stable 3.7.13.4-4+b1 armhf
libgnuradio-fcd3.7.13/stable 3.7.13.4-4+b1 armhf
libgnuradio-fcdproplus3.7.11/stable 3.7.25.4b6464b-5+b2 armhf
libgnuradio-fec3.7.13/stable 3.7.13.4-4+b1 armhf
libgnuradio-fft3.7.13/stable 3.7.13.4-4+b1 armhf
libgnuradio-filter3.7.13/stable 3.7.13.4-4+b1 armhf
libgnuradio-fosphor3.7.12/stable 3.7.0.2.7b6b996-3+b1 armhf
libgnuradio-hpsdr1.2/stable 1.2-1+b2 armhf
libgnuradio-iio0/stable 0.3-4 armhf
libgnuradio-iqbalance3.7.11/stable 0.37.2-11+b2 armhf
libgnuradio-limesdr0/stable 0.9~beta-1 armhf
libgnuradio-noaa3.7.13/stable 3.7.13.4-4+b1 armhf
libgnuradio-osmosdr0.1.4/stable 0.1.4-14+b7 armhf
libgnuradio-pager3.7.13/stable 3.7.13.4-4+b1 armhf
libgnuradio-pmt3.7.13/stable 3.7.13.4-4+b1 armhf
libgnuradio-qtgui3.7.13/stable 3.7.13.4-4+b1 armhf
libgnuradio-radar3.7/stable 0.0.0.20180308-1+b2 armhf
libgnuradio-rds3.7.11/stable 3.7.0.2.a542331-2+b2 armhf
libgnuradio-runtime3.7.13/stable 3.7.13.4-4+b1 armhf
libgnuradio-trellis3.7.13/stable 3.7.13.4-4+b1 armhf
libgnuradio-uhd3.7.13/stable 3.7.13.4-4+b1 armhf
libgnuradio-video-sdl3.7.13/stable 3.7.13.4-4+b1 armhf
libgnuradio-vocoder3.7.13/stable 3.7.13.4-4+b1 armhf
libgnuradio-wavelet3.7.13/stable 3.7.13.4-4+b1 armhf
libgnuradio-wxgui3.7.13/stable 3.7.13.4-4+b1 armhf
libgnuradio-zeromq3.7.13/stable 3.7.13.4-4+b1 armhf

op25 works great on a boot drive with gnuradio 3.7

Since the gnluradio 3.8 packages are not present, I cannot use ./install.sh
Here is the installation procedure that I used.
 
cd ~
git clone https://github.com/boatbod/op25
cd op25
sudo sed -i -- 's/^# *deb-src/deb-src/' /etc/apt/sources.list
# Tell op25 to use python3
echo "/usr/bin/python3" > op25/gr-op25_repeater/apps/op25_python
mkdir build
cd build
cmake ../
make -j4
sudo make install
sudo ldconfig

boatbod, 2020-12-26:
As you noted, the install.sh script is going to be looking at the available package version of gnuradio to determine how the install should proceed, but you are using a later gr built from sources which the script is not written to handle.

When you first start the cmake what do you see in the log message "GnuRadio version ..."?  It is this determination of gnuradio version which drives compile-time decisions for link libraries and appropriate version of python to use during swig compilation.  Even if cmake is detecting gr-3.8 correctly it is very conceivable you may be missing gr-3.8 build dependencies which would normally have been installed using the "sudo apt-get builddep gnuradio" command. Unfortunately that command won't work because it will install build dependencies for gr-3.7.

Suggest running through the "cmake ../" step and "make" steps while logging everything to file so you can review it to see what failed.
e.g.
    cd build
    cmake ../ >& cmake.log
    make >& make.log

srs4511351, 2020-12-27:
Is there a list prerequisites for op25? I was not able to find one.
I have gone over the GNU Radio prerequisites, but there are many, so maybe I missed one? My other GNU Radio applications work.
There doesn't seem to be an option to attach the log files, but I could paste them in another message. cmake.log is large.

I hope that you can see what is wrong here.

The cmake log shows that it could not find Boost::thread, Boost::program_options and Boost::regex as included in my first message.

It also says
--   Found gnuradio-runtime, version 3.8.2
(later)
-- GnuRadio version 3.8.2.0
-- Configuring for Python 3

-- Found SWIG version 3.0.12.

-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   filesystem
--   system

Relevant Boost packages installed are:
libboost-thread-dev/stable,now 1.67.0.1+b1 armhf [installed,automatic]
libboost-thread1.67-dev/stable,now 1.67.0-13+deb10u1 armhf [installed,automatic]
libboost-thread1.67.0/stable,now 1.67.0-13+deb10u1 armhf [installed,automatic]
libboost-program-options-dev/stable,now 1.67.0.1+b1 armhf [installed,automatic]
libboost-program-options1.67-dev/stable,now 1.67.0-13+deb10u1 armhf [installed,automatic]
libboost-program-options1.67.0/stable,now 1.67.0-13+deb10u1 armhf [installed,automatic]
libboost-regex-dev/stable,now 1.67.0.1+b1 armhf [installed,automatic]
llibboost-regex1.67-dev/stable,now 1.67.0-13+deb10u1 armhf [installed,automatic]
libboost-regex1.67.0/stable,now 1.67.0-13+deb10u1 armhf [installed,automatic]

The make log ends with:
/usr/bin/ld: cannot find -lBoost::program_options
/usr/bin/ld: cannot find -lBoost::regex
/usr/bin/ld: cannot find -lBoost::thread
collect2: error: ld returned 1 exit status
make[2]: *** [op25/gr-op25/lib/CMakeFiles/gnuradio-op25.dir/build.make:485: op25/gr-op25/lib/libgnuradio-op25.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:179: op25/gr-op25/lib/CMakeFiles/gnuradio-op25.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

srs4511351, 2020-12-27:
The log from gnuradio cmake shows a more complete list of Boost libraries.
-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   date_time
--   program_options
--   filesystem
--   system
--   regex
--   thread
--   unit_test_framework
--   chrono
--   atomic

srs4511351, 2020-12-28:
I now have op25 running on a Raspberry Pi 4 that has gnuradio 3.8 and an SDRplay RSP1A.
I installed a patched gr-osmosdr from osmocom instead of the version on https://github.com/fventuri/gr-osmosdr
I also upgraded cmake from 3.13.4 to 3.19.2.
I think the cmake upgrade is what caused cmake to be successful.
I will be documenting this on http://web.eecs.umich.edu/~srs/tech.html

boatbod, 2020-12-28:
Glad you got it working

closed 2020-12-28 by boatbod