boatbod/op25 #94: First line entry in TGID file is not getting imported

marklt1, 2021-01-04:
**I'll have a TGID file with the following entries and the first line doesn't get imported. Whatever entry I put on the first line is skipped.** 

13548	RHFD
13624	Beachwood12 PepperPike62 PD
13638	Moreland Hills School
13644	RHPD MARCS	1
13650	WEB Disp

_To work around this issue, I have put a blank line first and then start my entries on the second line._

**Below is the resulting log:**
using ALSA sound system
audio device: default
Listening on 127.0.0.1:23193
01/04/21 17:01:20.084883 [trunking_0] Initializing P25 system
01/04/21 17:01:20.084993 [trunking_0] reading system tgid_tags_file: trs_6643.tsv
01/04/21 17:01:20.086827 [trunking_0] setting tgid(13624), prio(3), tag(Beachwood12 PepperPike62 PD)
01/04/21 17:01:20.087079 [trunking_0] setting tgid(13638), prio(3), tag(Moreland Hills School)
01/04/21 17:01:20.087156 [trunking_0] setting tgid(13644), prio(1), tag(RHPD MARCS)
01/04/21 17:01:20.087223 [trunking_0] setting tgid(13650), prio(3), tag(WEB Disp)

boatbod, 2021-01-05:
Works fine here.
```
7101    MdTAP FMHrb
7102    MdTAP PtKey
7103    MdTAP BWI
7104    MdTAP ICC
7105    MdTAP Hatem
7106    MdTAP BayBr
...
```
```
01/04/21 19:32:59.499668 [FiRST Talbot] Initializing P25 system
01/04/21 19:32:59.499704 [FiRST Talbot] reading system tgid_tags_file: first.tsv
01/04/21 19:32:59.500214 [FiRST Talbot] setting tgid(7101), prio(3), tag(MdTAP FMHrb)
01/04/21 19:32:59.500240 [FiRST Talbot] setting tgid(7102), prio(3), tag(MdTAP PtKey)
01/04/21 19:32:59.500255 [FiRST Talbot] setting tgid(7103), prio(3), tag(MdTAP BWI)
01/04/21 19:32:59.500266 [FiRST Talbot] setting tgid(7104), prio(3), tag(MdTAP ICC)
01/04/21 19:32:59.500275 [FiRST Talbot] setting tgid(7105), prio(3), tag(MdTAP Hatem)
01/04/21 19:32:59.500289 [FiRST Talbot] setting tgid(7106), prio(3), tag(MdTAP BayBr)
```
Have you checked for invalid format?  i.e. spaces instead of tabs
If a line is bad it will be skipped.

closed 2021-01-05 by marklt1

boatbod, 2021-01-06:
I have made changes to the python routines for both trunk-tags.tsv and trunk.tsv to strip out the UTF8 BOM if present.

marklt1, 2021-01-06:
Works perfect now! 
As always, thanks for what you do with this project.