boatbod/op25 #95: Audio problem with multi_rx.py

srs4511351, 2021-01-06:
I think I have multi_rx.py working enough to receive amateur radio DMR, but I can't get any audio.
I have rx.py working on the public safety trunked system using either external or internal audio.
I have a Raspberry Pi4 with the SDRplay RSP1A SDR.

I am running it in 3 terminals:
cd ~/op25/op25/gr-op25_repeater/apps
#1
./multi_rx.py -c dmr_cfg.json -v 9 2> stderr.2

#2
./audio.py -2

#3
tail -f stderr.2

----dmr_cfg.json----
{
    "channels": [
        {
            "device": "driver=sdrplay,soapy=0",
            "demod_type": "fsk4", 
            "destination": "udp://127.0.0.1:23456", 
            "excess_bw": 0.2, 
            "filter_type": "rrc", 
            "frequency": 443500000, 
            "if_rate": 24000, 
            "name": "CC", 
            "plot": "fft", 
            "symbol_rate": 4800
        }
    ], 
    "devices": [
        {
            "args": "driver=sdrplay,soapy=0", 
            "frequency": 443500000, 
            "gains": "IFGR:28,RFGR:1", 
            "name": "RSP1A", 
            "offset": 0, 
            "ppm": 0, 
            "rate": 500000, 
            "tunable": false
        }
    ]
}

With debug set to 9, when audio was being received there were many messages like this:
01/04/21 23:21:19.192453 [-1] AMBE e4 1c 27 60 9f 18 00 errs 5 err_rate 0.005320
01/05/21 18:28:06.281583 [-1] AMBE f8 03 73 ce cc 09 80 errs 0 err_rate 0.019816
01/05/21 18:28:06.281613 [-1] AMBE f8 03 73 c8 cc af 80 errs 3 err_rate 0.022018

With debug set to 10 (signal received, no audio transmitted):
01/05/21 21:39:18.747864 [-1] SLCO=0x0, NULL MSG
01/05/21 21:39:18.748071 [-1] Slot(1), CC(1), Data Type=9, gly_errs=0
01/05/21 21:39:18.748102 [-1] Slot(1), CC(1), IDLE
01/05/21 21:39:18.773760 [-1] Slot(0), CC(1), Data Type=9, gly_errs=0
01/05/21 21:39:18.773826 [-1] Slot(0), CC(1), IDLE
01/05/21 21:39:19.128434 [-1] DMR: sync expiry, symbol 92640

In the audio terminal, audio.py -2 crashed as soon as audio was received.
$ ./audio.py -2
Using Python /usr/bin/python3
using ALSA sound system
audio device: default
Listening on 0.0.0.0:23456
Traceback (most recent call last):
  File "./audio.py", line 74, in <module>
    audio_handler.run()
  File "/home/pi/op25/op25/gr-op25_repeater/apps/sockaudio.py", line 467, in run
    data_b = self.scale(data_b)
  File "/home/pi/op25/op25/gr-op25_repeater/apps/sockaudio.py", line 477, in scale
    arr = np.array(np.frombuffer(data, dtype=np.int16), dtype=np.float32)
AttributeError: 'str' object has no attribute '__buffer__'
---
Witout the -2 switch, it didn't crash immediately:
$ ./audio.py
Using Python /usr/bin/python3
using ALSA sound system
audio device: default
Listening on 0.0.0.0:23456
01/06/21 14:52:06.249303 PCM underrun
01/06/21 14:52:11.536937 PCM underrun
01/06/21 14:52:16.382240 PCM underrun
01/06/21 14:52:20.742466 PCM underrun
01/06/21 14:52:30.749810 PCM underrun
01/06/21 14:52:37.162530 PCM underrun
Traceback (most recent call last):
  File "./audio.py", line 74, in <module>
    audio_handler.run()
  File "/home/pi/op25/op25/gr-op25_repeater/apps/sockaudio.py", line 461, in run
    data_a = self.scale(data_a)
  File "/home/pi/op25/op25/gr-op25_repeater/apps/sockaudio.py", line 477, in scale
    arr = np.array(np.frombuffer(data, dtype=np.int16), dtype=np.float32)
AttributeError: 'str' object has no attribute '__buffer__'

----Steve

boatbod, 2021-01-07:
Are you running the latest code?  I worked on some DMR no-audio and sockaudio.py crashing issues in this very location just last weekend (1/2/21).  Might want to "git pull" just to make sure.

Assuming you are up to date, please can you add the parameter string `   "raw_output": capture.bin,    ` in the channel section of your config file, then run the multi_rx app until you see some AMBE codewords produced.  Stop the app and email me both the "capture.bin" file and a copy of your stderr.2 log.   gnorbury@bondcar.com

Thanks

boatbod, 2021-01-25:
Closed due to no response to followup questions.

closed 2021-01-25 by boatbod

srs4511351, 2021-01-25:
I want to apologize for not answering your comment.
The git pull fixed the problem, but if you want, I can still make the capture for you.

Otherwise, please feel free to close this again.

----Steve