boatbod/op25 #96: trouble running with .tsv file

kmarundee, 2021-01-07:
If I run this:
./rx.py --args 'rtl' -N 'LNA:47' -S 2500000 -x 2 -f 774631250 -o 17e3 -q -1
It work fine an I can receive the NAC

If I run this: 
./rx.py --args 'rtl' -N 'LNA:47' -S 2500000 -x 2 -f 774631250 -o 17e3 -q 0 -T trunk.tsv -w -U 2> stderr-stream0.
it shows that it is on 774.63125 but does not show the frequency up at the top only 0.000000/0.000000 and the spectrum doesn't show the "spike" of the control channel. also removing " -f 774631250" from the command line results in the same. so i feel it is reading the trunk.tsv file

trunk.tsv = 
`Sysname	Control Channel List	Offset	NAC	Modulation	TGID Tags File	Whitelist	Blacklist	Center Frequency
P25 SYSTEM 	744.631250 	0	0x388 	cqpsk	work.tsv`

work.tsv = 
 `856	Dispatch a
866	Dispatch b
`

Any help or suggestions would be appreciated.

boatbod, 2021-01-07:
Is the trunk.tsv file correctly formatted?  One and only one tab between each parameter, which in my example below are delimited with quotes.  The quotes are actually optional, but make it easier to differentiate.
`"Sysname"       "Control Channel List"  "Offset"        "NAC"   "Modulation"    "TGID Tags File"
        "Whitelist"     "Blacklist"     "Center Frequency"
"P25 SYSTEM"    "876.54320"     "0"     "0"     "cqpsk" ""      ""      ""      ""`

kmarundee, 2021-01-07:
Yes I have double checked tab spacing. 

Thank you, 
Kevin Marundee

boatbod, 2021-01-07:
Add the command line parameter "-v 1" and re-run.  Do you see anything relevant in the stderr.stream0 log?

kmarundee, 2021-01-08:
I ran it with the -v an the stderr.stream0 showed the frequency timed out.  Turns out that I entered the frequency wrong.  Works great now. Thank you for your help. 

Kevin

closed 2021-01-13 by boatbod