boatbod/op25 #100: Failed to install op25 on kali linux

khanfar, 2021-01-30:
Hi , when i try to install op25 on kali linux nethunter using this command  ./install.sh i got this error in attached picture below , please any help 
[image]

khanfar, 2021-01-30:
(( This is CMake Error.log ))

[image]

khanfar, 2021-01-30:
(( This is CMake Output.log  ))

[image]

boatbod, 2021-02-06:
From the error with FindTHRIFT.cmake I assume your version of kali is running gnuradio 3.8.2?
For reasons best known to the GR developers there were changes made in the gnuradio cmake scripts which force the incorporation of the "THRIFT" module, but for some reason FindTHRIFT is not properly including GrPython so the script fails.

I haven't yet found a good solution that doesn't break earlier versions, but as a workaround you can edit the FindTHRIFT.cmake file (found under the /usr/lib hierachy)  and place "include(GrPython)" on the first line then re-run the cmake and it should work.

boatbod, 2021-02-07:
I found and implemented a fix for the FindTHRIFT / GrPython cmake problem in gnuradio-3.8.2

closed 2021-02-07 by boatbod