boatbod/op25 #101: No Audio

voice06, 2021-01-31:
Running rx.py, previously it was able to handle audio just fine but recently it seems that has stopped. I see no errors in the console and when I check the volume control app I can see the ALSA Plugin for python3.8 popping in and out but there just seems to be no audio data making it from rx.py / sockaudio.py to ALSA, its the same when I set it to pulse as well.

boatbod, 2021-02-06:
Check the stderr.2 log file for any errors, and also look for any lines containing the keyword "audio".
Op25 opens the audio device at startup and leaves it open for the duration of the session, so it should show up in either alsamixer or pavucontrol.

Are you sure audio is being decoded?  If you have logging at "-v 1" or higher, you should see "voice update:" messages in the log whenever op25 tunes to a voice channel.

voice06, 2021-02-08:
Looks like audio data is being received and it connects with Pulse or ALSA
```gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
The RTL TCP server reports a R820T tuner with 29 RF and 0 IF gains.
gain: name: LNA range: start 0 stop 0 step 0
setting gain LNA to 49
supported sample rates 250000-2560000 step 24000
Using two-stage decimator for speed=960000, decim=10/4 if1=96000 if2=24000
op25_audio::open_socket(): enabled udp host(127.0.0.1), wireshark(23456), audio(23456)
p25_frame_assembler_impl: do_imbe[1], do_output[0], do_audio_output[1], do_phase2_tdma[1], do_nocrypt[1]
02/08/21 16:37:47.934379 Reading blacklist file
02/08/21 16:37:47.934431 added talkgroup 5803 from dupage_blacklist.tsv
02/08/21 16:37:47.934449 added talkgroup 34481 from dupage_blacklist.tsv
02/08/21 16:37:47.934458 added talkgroup 2617 from dupage_blacklist.tsv
02/08/21 16:37:47.934467 added talkgroup 2615 from dupage_blacklist.tsv
metadata update not enabled
using PulseAudio sound system
audio device: pulse
Listening on 127.0.0.1:23456
python version detected: 3.8.5 (default, Jul 28 2020, 12:59:40) 
[GCC 9.3.0]
02/08/21 16:37:48.611022 voice update:  tg(33858), freq(772456250), slot(-), prio(3)
02/08/21 16:37:48.670408 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:48.713351 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:48.756049 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:48.798733 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:48.849869 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:48.892666 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:48.909585 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:48.926820 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:49.003622 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:49.182668 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:49.361953 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:49.541087 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:50.036299 voice update:  tg(33858), freq(772456250), slot(-), prio(3)
02/08/21 16:37:50.087279 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:50.266454 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:50.283499 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:50.326179 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:50.368780 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:50.411481 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:50.462670 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:50.505292 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:50.548019 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:50.599320 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:50.641925 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:50.684463 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:50.727186 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:50.778435 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:50.821115 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:50.863843 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:50.906395 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:50.957617 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:51.000389 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:52.136044 voice update:  tg(2017), freq(770806250), slot(-), prio(3)
02/08/21 16:37:52.204006 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:52.382782 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:52.561857 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:52.741033 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:52.920276 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:53.099637 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:53.116443 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:53.159221 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:53.406898 voice update:  tg(2017), freq(770806250), slot(-), prio(3)
02/08/21 16:37:53.474862 [0] p25_framer::rx_sym() tuning error -1200
02/08/21 16:37:53.517585 [0] p25_framer::rx_sym() tuning error -1200
```
[image]
[image]

boatbod, 2021-02-09:
The persistent tuning error messages are likely of more significance than you may realize.  I think the tuning is far enough off that you may not be decoding much voice.  Please capture and post (or email) a screen-shot of the datascope plot.

All the traffic I see in the included log fragment is FDMA.  If this is a phase-1 only system it may possible that it is using C4FM modulation rather than the default DQPSK?  Again the datascope plot should reveal what's going on here too.

voice06, 2021-02-11:
Looks like DQPSK/C4FM was the issue, added the demod-type flag to use fsk4 and its got audio now and no tuning errors.

boatbod, 2021-02-11:
Glad to hear it's working.

closed 2021-02-11 by boatbod