boatbod/op25 #102: op25 fails to start on Raspberry PI and Ubuntu

srs4511351, 2021-02-05:
I had previously managed to install and run op25 on Raspberry Pi OS, but I am having problems with my new Ubuntu installation.
The other SDR applications that are installed are working.

Initially, the error was:
Using Python /usr/bin/python3
osmosdr source_c creation failure
Traceback (most recent call last):
  File "./rx.py", line 1068, in <module>
    rx = rx_main()
  File "./rx.py", line 975, in __init__
    self.tb = p25_rx_block(self.options)
  File "./rx.py", line 169, in __init__
    gain_names = self.src.get_gain_names()
AttributeError: 'NoneType' object has no attribute 'get_gain_names'
---
After reinstalling op25, now I just see:
osmosdr source_c creation failure

I found this error in cmake output, but MPIR is mostly the same as gmp.
-- Checking for module 'gmp'
--   Found gmp, version 6.2.0
-- Found GMP: /usr/lib/arm-linux-gnueabihf/libgmpxx.so  
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- Found MPLIB: /usr/lib/arm-linux-gnueabihf/libgmpxx.so

However, cmake produced a lot of long unsigned int and unsigned integer errors

I reinstalled gr-osmosdr.

I have installed
Ubuntu 20.04.2
Kernel: Linux 5.4.0-1028-raspi
gnuradio 3.8.1.0~rc1-2build2

I have attached a file with the output from cmake, make and sudo make install
[op25 install logs Ubuntu 2-5-2021.txt](https://github.com/boatbod/op25/files/5935438/op25.install.logs.Ubuntu.2-5-2021.txt)

boatbod, 2021-02-05:
Nothing wrong with the build, but op25 cannot open the SDR device.
[quote]
osmosdr source_c creation failure
[/quote]

Either you didn't get the rtl drivers blacklisted properly or (if this is a virtual machine) you didn't tell the vm host to pass the USB device on to the guest.

closed 2021-02-05 by boatbod

srs4511351, 2021-02-05:
Thanks for the clarification. 
it is not an RTL or a virtual machine. 
Both of my SDRs work on 4 other applications, some using gr-osmosdr.
I have a SDRplay RSP1, and some say to blacklist the sdr_msi3101, msi001 and msi2500 but that has never made any difference when I tried it. 
The other is a hackrf. I had both working on Raspberry Pi OS.

boatbod, 2021-02-06:
I'm not overly familiar with the "args" parameter required for RSP1, but whatever you use with gqrx or similar gnuradio-based apps should work the same with op25.

srs4511351, 2021-02-06:
I added the driver blacklists, but it still doesn't work. However, I found something that may be important.
I ran multi_rx.py and looked in stderr.2:
Using Python /usr/bin/python3
Traceback (most recent call last):
  File "./multi_rx.py", line 52, in <module>
    import osmosdr
ModuleNotFoundError: No module named 'osmosdr'

osmosdr is installed and works with other applications, but they must not be using python.
When I try it at the python3 prompt on my Ubuntu system, it also fails, but it works on my Raspberry Pi OS system where op25 works.
The big difference with the Ubuntu system that doesn't work, is that I used the packager for gnuradio, which was actually installed by the packager when I installed UHD.
GNU Radio version is 3.8.1.0, but the version is 3.8.2.0 on the installation that works.

----Steve

boatbod, 2021-02-06:
I expect osmosdr has ben installed under python2 but not python3.   

To confirm this, see what happens if you run the following two commands:
   python3 -c "import osmosdr"
   python2 -c "import osmosdr"

If the module is installed the command will complete with no error.  If it's not, it will error as follows:
   Traceback (most recent call last):
     File "<string>", line 1, in <module>
   ImportError: No module named osmosdr

You might try uninstalling/reinstalling the gr-osmocom package, but to be honest I'm not sure how it makes the determination of where it is going to install itself.  The other option would be to pull the source and build/install it yourself.  
   apt-get source gr-osmosdr
   cd gr-osmosdr-*
   mkdir build
   cd build
   cmake ../
   make
   sudo make install

srs4511351, 2021-02-06:
> File "", line 1, in

Thanks for the python commands.
It failed both of them.

I left out this step for the packager installed gnuradio, but adding this
fixed the python3 import!
export PYTHONPATH=/usr/local/lib/python3/dist-packages:$PYTHONPATH
Apparently, I can't trust them to do it all.

I will try installing op25 later, as I have restarted my install from
scratch.

----Steve

boatbod, 2021-02-06:
Hmmmm, so is gr-osmosdr installed at all?  If not, that would be rather odd since it is specifically listed in the install.sh script.  What happens if you run this?
   apt list | grep osmosdr

If gr-osmosdr does not show as "installed", you could try running this:
   sudo apt-get install gr-osmosdr

srs4511351, 2021-02-07:
I think you misunderstood my last post.
When I added "export
PYTHONPATH=/usr/local/lib/python3/dist-packages:$PYTHONPATH" to .profile
and rebooted, python will successfully import osmosdr.
The version on my system of gr-osmosdr is an OOT fork for SDRplay. So apt
will not show it as installed.
The osmocom version does not recognize the latest SDRplay driver. It does
work if I replace a few files from the SDRplay fork.

As of now, it is working on my Ubuntu system. I'm listening to it now.

----Steve

boatbod, 2021-02-07:
I found and implemented a fix for the FindTHRIFT / GrPython cmake problem in gnuradio-3.8.2

Waymorehoss, 2024-08-28:
> I think you misunderstood my last post. When I added "export PYTHONPATH=/usr/local/lib/python3/dist-packages:$PYTHONPATH" to .profile and rebooted, python will successfully import osmosdr. The [...]

how did you get sdrplay to work on op25?

srs4511351, 2024-08-28:
I used a different branch of osmosdr. The current official branch looks for SDRplay API 2.x, where it has been 3.x for years.
My procedure is on this page:
https://radiosrs.net/installing_drivers_sdrplay_and_HackRF_Bookworm_RPI_OS-64.html

It is targeted at Raspberry Pi OS, but it should be relevant to any recent linux. What CPU and OS are you using?
What version of GNU Radio do you have?
Here are the terminal commands it install it for **GNU Radio 3.9 or later**:
```
cd ~/
git clone https://github.com/fventuri/gr-osmosdr.git
cd gr-osmosdr/
mkdir build
cd build
cmake .. -DENABLE_NONFREE=TRUE

# Under Gnuradio enabled components in the terminal text, check that SDRplay RSP (NONFREE), SoapySDR, HackRF and  Osmocom RTLSDR support are listed.

make -j4
sudo make install
sudo ldconfig
```
There are a few problems with this release, but it works.
If you get it to work, look at 
https://github.com/fventuri/gr-osmosdr/issues/2
and
https://github.com/gqrx-sdr/gqrx/issues/1345

----Steve

Waymorehoss, 2024-08-30:
> [fventuri/gr-osmosdr#2](https://github.com/fventuri/gr-osmosdr/issues/2)
> [gqrx-sdr/gqrx#1345](https://github.com/gqrx-sdr/gqrx/issues/1345)

what is the command you use to start op25 with your sdrplay device? BTW i am using rasbian os
S22 ultra

Sent with [Proton Mail](https://proton.me/) secure email.

srs4511351, 2024-08-30:
I am not familiar with the differences between raspbian os S22 ultra and Raspberry Pi OS Bookworm. Is is an old version?
What version of GNU Radio do you have? Type gnuradio-config-info -v

My procedures are on https://radiosrs.net/tech.html
For older 32 bit Raspberry Pi installations, I have a section titled Raspberry Pi OS Bullseye 32-bit installation
There is a page for op25 and how to configure it for your system.
This command calls your configuration, which depends on the system that you have in your area.
Here is my command
```
# For SDRplay:
cd ~/op25/op25/gr-op25_repeater/apps
./rx.py --args 'driver=sdrplay,soapy=0' -N 'IFGR:28,RFGR:1' -S 2500000 -x 1 -f 855.2875e6 -o 1e3 -q -1 -d 100 -T michigantrunk.tsv -V -2 -O pulse -U 2> stderr-stream0.2
```

Waymorehoss, 2024-08-30:
I'm sorry s22 ultra is just my phone that this email came from...i am using this OS https://github.com/luigifcruz/pisdr-image it is a 64bit version of rasbian os but it has all the software for software defined radio already built into it. i have op25 up and running great with my RTL-SDR v4 dongle it's getting it working with my RSPdx that i'm trying to do. do i need gnu radio installed i am not familar with that oh and i am quite new to the hobby. lets say on a fresh install of rasbian os 64bit and a fresh install of op25 what steps would you take from there to make op25 recognize your RSPdx?

S22 ultra

Sent with [Proton Mail](https://proton.me/) secure email.

srs4511351, 2024-08-30:
Try the command I gave in my last post, but substitute your configuration
files.
I have a RSP1A, but it should work.
Tell me what happens.

I will have to review the op25 requirements, but I think it uses GNU Radio.

My site also has pre installed images with a large list of software.

----Steve

> I am not familiar with the differences between raspbian os S22 ultra and
Raspberry Pi OS Bookworm. Is is an old version?
> What version of GNU Radio do you have? Type gnuradio-config-info -v
>
> My procedures are on https://radiosrs.net/tech.html
> For older 32 bit Raspberry Pi installations, I have a section titled
Raspberry Pi OS Bullseye 32-bit installation
> There is a page for op25 and how to configure it for your system.
> This command calls your configuration, which depends on the system that
you have in your area.
> Here is my command
>
> # For SDRplay:
> cd ~/op25/op25/gr-op25_repeater/apps
> ./rx.py --args 'driver=sdrplay,soapy=0' -N 'IFGR:28,RFGR:1' -S 2500000 -x
1 -f 855.2875e6 -o 1e3 -q -1 -d 100 -T michigantrunk.tsv -V -2 -O pulse -U
2> stderr-stream0.2
https://github.com/boatbod/op25/issues/102#issuecomment-2322189546), or
[unsubscribe](
https://github.com/
).

<https://github.com/boatbod/op25/issues/102#issuecomment-2322399595>, or
unsubscribe

srs4511351, 2024-08-31:
With the image that you installed, the problem is probably the gr-osmosdr version that they used.
I looked at pisdr-image, but I cannot find any version details.

You can find the SDRplay API version by typing:
SoapySDRUtil --probe="driver=sdrplay"

The GNU Radio version can be found by typing"
gnuradio-config-info -v

The original gr-osmosdr from osmocom does not support the current SDRplay API.

This is displayed when you start op25, showing that is is using gnuradio and gr-osmosdr
```
Using Python /usr/bin/python3
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.5.1
built-in source types: file rtl rtl_tcp uhd sdrplay hackrf rfspace soapy redpitaya xtrx
...
```
Does your SDRplay work with gqrx?

srs4511351, 2024-08-31:
I installed the latest image from https://github.com/luigifcruz/pisdr-image
It has the SDRplay API driver version 3.07, which is good for all but the two new devices released this year.
It has the gr-osmosdr from osmocom, which **does not work** with v 3.07 of the SDRplay API driver that is installed.
It has an early pre release version of GNU Radio 3.10.
gqrx recognized the SoapySDR driver for my SDRplay, but the gr-osmosdr device is not present, as is is on my system.

In order to get op25 working, you will have to reinstall gr-osmosdr with the version found here:
https://github.com/fventuri/gr-osmosdr
You may also have to reinstall applications installed after the original gr-osmosdr was installed.

Complete instructions to install a system from scratch are available here:
https://radiosrs.net/tech.html

There is also a 64-bit Raspberry Pi OS Debian Bookworm disk image with all of the software on it.
A [list of software](https://radiosrs.net/RPI-OS_disk_img_contents.html#software_list) is available.

srs4511351, 2024-08-31:
I just realized that the command I use to start op25 specifies SoapySDR instead of the gr-osmosdr driver, which may work for you. If not use the device string specification found in GQRX, in the settings window.

When I get time, I’ll try running op25 on that system.

Waymorehoss, 2024-08-31:
Oh WOW is all i can say i installed the raspberry pi bookworm 64bit image that you have on your website and it has so much stuff already installed and it's so easy to use i love it right now i have my RSPdx working with op25 i just put that command you sent me in and modified it alittle but it's working i can't believe it. this is all so far outside my skill to setup but this is just working out of the box i like it!. i see in the op25 folder that you already have your stuff put in such as michigantrunk.tsv i'm use to using just regular op25 so all those extra files u have i'm not use to seeing. why in the michagan.tsv file do you have all those freqs? i'm use to just using one if i want to monitor two control channels do i just put those in? and under "Other" in the start menu i see stuff like Rdio Scanner-P25 and Trunk Recorder-SDRPlay and Start P25 scanner and when i click on those 3 terminals come up but thats it nothing else runs. Like when i click Start P25 scanner this shows up " i'll attach a pic" does op25 work as a scanner to? right now i'm having to run dsdplus fastlane on a separate pc just so i can have the scan function i listen to 3 freqs two are p25p1 but you don't have to lock onto a control channel and the other is NXDN48 CB ....Anyway's i really like what you have put together here it has made things so much easier already. my area trunking is all P25 phase 2 ...which files do i edit in the op25 apps folder for sdrplay? i tried editing the michigantrunk.tsv and michigan.tsv i changed the names to my county and changed everything inside to but now if i run the whole command it won't open but it still opens if i just run ./rx.py --args 'driver=sdrplay,soapy=0' -N 'IFGR:28,RFGR:1' -S 2500000 -x 1 -f 855.2875e6 -o 1e3 -q -1

Sent with [Proton Mail](https://proton.me/) secure email.

Waymorehoss, 2024-08-31:
S22 ultra

Sent with [Proton Mail](https://proton.me/) secure email.

Waymorehoss, 2024-08-31:
do i need to tell op25 which antenna to connect to? like on my rspdx i have 3 antenna connectors and i know on stuff like sdrtrunk and dsdplus i have to tell it what antenna to use. right now i have it working i couldn't figure out which trunk.tsv file was used for sdrplay i eventually tried them all and the only one that seems to work is the umichtrunk.tsv one here is a screenshot of what it looks like now. can i monitor two control channels at once? i know with my rtl-sdr dongle i could put two control channels in trunk.tsv and it would rotate control channels i tried that with the sdrplay but it wouldn't let the command run.

S22 ultra

Sent with [Proton Mail](https://proton.me/) secure email.

srs4511351, 2024-08-31:
Wow, that's lot of information.
I will try to answer your questions, but if I miss something, ask again.
I suggest that you **read my op25 configuration page**. 
op25 is a P25 Phase 2 scanner. You don't need DSD Plus.

You need the NAC number, which is in the instructions. It should be on 
https://www.radioreference.com/apps/db/

The list of frequencies in michagantrunk.tsv are the control channels. You need all of them because they occasionally change.
When using control channels, it finds the activity on all of the frequencies on your system. It will then list them and the talk groups that are active.
The channel or talkgroup that you want to hear could be on any of those frequencies.
radioreference.com has all of this information. The other file, michigan.tsv contains the TGID or talkgroups. With this file, op25 will display a description of the talkgroup, or what entity is talking.
If you can tell me what system you are using, I can download the TGID file for you.
If you are in Michigan, you already have it, however I see some unidentified talkgroups, so I need to update it.

You do need the whole command that I sent to spend it the trunking configuration. Just change it to your configuration files.
op25 is trunked, so you do need all of the control channels and system ID/NAC. The entity that you want to listen to could be on any of the frequencies.

I will have to research the switch for the antenna selection.
Only 1 control channel is used at a time. Sometimes it changes, so you need to specify them all.

If you want to monitor two systems at once, you need to  run 2 instances of op25 with separate SDRs.

srs4511351, 2024-08-31:
Rdio Scanner-P25 and Trunk Recorder is another P25 scanner. It can be run over the Internet.
SDRthunk is a P25 scanner written in the java language and runs poorly on the Raspberry Pi.
There are 3 terminals in op25 because one is the errors and system messages, one is for liquidsoap, which processes the audio, and one is op25 which is where you see who is talking, etc.
I am unable to see your pictures. You may have to click the View it on Github link and do it there, or we could hook up on email.

Waymorehoss, 2024-09-01:
Ok i have op25 working with the rspdx and i got the gui web interface going as well. i don't think their is a antenna selection because i tried all three antenna ports and each one works with the same command. i can't use sdrtrunk with my rspdx because the freqs i need are outside the 10mb bandwidth so i would have to add a dongle with it. 

I understand what your saying about opening two instances of op25 to listen to two different control channels.  i have a radio reference subsciption i downloaded the talkgroups and put them in, in the op25 folder i seen 3 files michigantrunk.tsv and umichigan.tsv and trunk.tsv why are they 3 different files i know when i download just op25 before i only had 1 file trunk.tsv.

I really like this OS you have setup it's made things alot easier i was looking on the website for a .iso version of the raspberry pi 64bit os but i only see a .img i would like to install this copy of rasbian on a pc and i think i need a .iso to do that.

What about scanning like dsdplus fastlane does is their a way on linux based systems to scan digital but without using dsdplus at all? i have my P25 which is state police and i have NXDN48 conv and two p25 phase 1 conv channels that i scan thorough using dsdplus fastlane but i have to keep a pc running windows to do that and i'm on linux all the time, i'm just curious if there are any other programs that can do that. i was gonna look into dsd-fme i was reading that it has a scanning function but i don't know anything about it yet. Thank you again for all you have put together it really is awesome.

Waymorehoss, 2024-09-01:
do you know how to activate the bias tee in the op25 command? where do i put it

srs4511351, 2024-09-01:
op25 and SDRtrunk both change the center frequency of your SDR to cover the frequency that you are listening to. However, SDRtrunk needs more CPU power than  the Pi can provide. op25 only uses a few percent of your CPU.
Trunk Recorder does not seem to be able to do that and will report that it can't tune certain frequencies.

It still sounds like you do not have op25 set up to scan. You do not need DSDplus.
The image came with op25 configured to scan digital for my system and it works. You only need to change the talkgroups file and the trunk.tsv for your system as specified in my instructions.
The only frequencies that you should specify are the complete list of control channels. Only one control channel will be in use by the system you are monitoring at any given time.
See: https://radiosrs.net/installing_op25.html

trunk.tsv is as follows:
Change the NAC to the one specified for your system on radioreference.
Use your control channels
Sysname | Control Channel List | Offset | NAC | Modulation | TGID Tags File | Whitelist | Blacklist | Center Frequency
-- | -- | -- | -- | -- | -- | -- | -- | --
Your system name | 855.2875,852.8625,853.250,855.0875 | 0 | 0x799 | cqpsk | talkgroups.tsv |   |   |  

The file will look like this:
```
"Sysname"	"Control Channel List"	"Offset"	"NAC"	"Modulation"	"TGID Tags File"	"Whitelist"	"Blacklist"	"Center Frequency"
"Your system name"	" 855.2875,852.8625,853.250,855.0875"	"0"	"0x799"	"cqpsk"	"talkgroups.tsv"	""	""	""
```
The  command to turn on biasT is in the command below:
```
./rx.py --args 'driver=sdrplay,soapy=0,biasT_ctrl=true' -N 'IFGR:28,RFGR:1' -S 2500000 -x 1 -f 855.2875e6 -o 1e3 -q -1 -d 100 -T michigantrunk.tsv -V -2 -O pulse -U 2> stderr-stream0.2
biasT_ctrl=true turns it on
biasT_ctrl=false turns it off
```
For the antenna selection, I need to know something from your system. I do not have the rspdx.
Paste the command below at the command prompt. Copy the text and paste in into your message. Highlight the text with the mouse. You will have to scroll the terminal. Right-click and select copy)
`SoapySDRUtil --probe="driver=sdrplay"`

I don't know if op25 will decode NXDN48. Start a ticket and ask.
I don't know how to get op25 to scan the police trunking AND these other unrelated frequencies.

srs4511351, 2024-09-01:
Here's what op25 looks like when decoding and scanning talkgroups.
[image]

srs4511351, 2024-09-01:
SDRangel that is installed on your system has a DSD Demodulator. Give it a try for the NXDN48 signal. You will have to play with the settings.
There are signals in my area. I'll try some.

It also has a scanner, but no P25 trunking.

SDRangel can track satellites and apply doppler correction, including weather satellites and the International Space Station. I've seen slow scan TV from it.

srs4511351, 2024-09-01:
I found the SoapySDRUtil  info online and it wasn't much help.
Does the rspdx have lights that tell you which antenna is selected?
From stuff I found, I have suggestions for you to try, things like:
antenna=B
Antenna=B
etc.

./rx.py --args 'driver=sdrplay,soapy=0,biasT_ctrl=true,**antenna=B**' -N 'IFGR:28,RFGR:1' -S 2500000 -x 1 -f 855.2875e6 -o 1e3 -q -1 -d 100 -T michigantrunk.tsv -V -2 -O pulse -U 2> stderr-stream0.2

If you need more gain, try changing RFGR:1 to RFGR:0
Increasing IFGR:28 to something like IFGR:48 may help with overload.
You can look at the signal plots to see what it looks like.

Waymorehoss, 2024-09-01:
No the rspdx doesn't have lights that tell the antenna instead i physically have to connect one of three antenna ports and they are labled.  the scanning i was talking about is scanning a set of freqs like 3 freqs at once and the scanner going through them one at a time really fast. is that what u are talking about with op25 scanning? or are u talking about just scanning multiple control channels at once? i will try out sdrangel

Waymorehoss, 2024-09-01:
so with sdrangel i can scan and decode digital without any other software? also i have a issue where my spectrum in Gnuplot will be doing fine and then it will just go all crazy and the signals will go dead . is that a overload? if i turn the gain to almost 0 it stops but then i can't hear the signal

Waymorehoss, 2024-09-01:
i have tested sdrangel i can not get any volume to work i have tried about everything. But basically i have op25 up and running i am monitoring 1 control channel i tried putting in another control channel with a different NAC underneath the first control channel in the trunk file but then the terminal won't open and start op25 i had got this to work before on a fresh install of op25 . i'm still confused what u mean by op25 being able to scan do you mean scanning through a set of freqs like dsdplus fastlane does? or do u mean just scanning different control channels for different sites? my biggest problem right now is i have to turn my gain all the way down because my spectrum becomes all crazy or scrambled like it's overloading but i can't pick up my signals by turning the gain down that far i have tried everything to stop this but i can't get it.

boatbod, 2024-09-01:
If you want to monitor two systems (2 control channels) you need to, either:
- configure multiple lines in trunk.tsv (rx.py)
- have multiple SDRs available (multi_rx.py)

In the former case (rx.py) the app will switch (scan) between the two 
systems.  In the latter case (multi_rx.py) the app will monitor both 
simultaneously.

Graham

Waymorehoss, 2024-09-01:
srs4511351 if you want to give me your email we can communicate on there.

S22 ultra

Sent with [Proton Mail](https://proton.me/) secure email.

srs4511351, 2024-09-01:
What I am talking about with op25 scanning is actually trunking where the control channel directs all of the traffic.
You need the list of control channels so op25 can find the **only one** that is currently in use.
op25 will choose one talkgroup for you to listen to. In this scenario, you can listen to any unencrypted conversation on the system. It looks like the picture I attached.

The other channels that you want to scan are in independent of and in addition to this.

The post from boatbod, the maintainer of op25, is very interesting. You can monitor multiple systems! That may open it up for you to scan your other P25 systems or individual frequencies together. I will be trying this.

I'm not sure from what you say that you actually mean multiple control channels.
The P25 system has one control channel and NXDN48 probably does not, at least the NXDN48 frequencies around here don't use a control channel.  A different type of thing.

So what I meant by scanning with op25 is just monitoring one system with its one control channel.

Waymorehoss, 2024-09-02:
Ok i do have op25 up and running with my RSPdx so far it seems to be working i still need to do some more testing currently i am running dsdplus fastlane on a laptop and op25 on my raspberry pi and sdrtrunk on my main pc i'm gonna test and see if one picks up better than the other.

My biggest issue right now is still the overload or whatever it is like i said my spectrum will be running fine then it will go all bazerk and go all over the place i have tested turning the gain down little by little but eventually i get to low of gain to pick anything up, the "overload" does stop once i go really low on the gain tho. i am running a LNA and FM filter in my setup so i took it off to see if that was it and it still done the same thing. does anyone else have this problem? or know what it could be? i'll attach two screenshots showing what i mean. the 1st pic is normal the 2nd is not.
[image]
[image]

Waymorehoss, 2024-09-02:
and what should i set my IFGR and RFGR to? i am use to the dongles and gain setting for those. i really don't even know what IFGR and RFGR stands for. like i said i'm learning i just got my RSPdx a week ago

srs4511351, 2024-09-02:
With SDRplay, 0 is the highest gain and a larger number is lower gain.
I find that best reception is with a high RFGR RF gains (0 or 1) and a lower IFGR IF gain. You can see that in the rx.py commands that I provided.

SDRangel has a wide variety of demodulators and features built in.
DSD Demod is included. It claims to work with NXDN!
Additionally, if your frequencies are all in the bandwidth of your SDR, say 10 MHz, you can open several demodulators, maybe one narrow FM, one digital voice, one AM frequency and play them all simultaneously, so it won't need to scan!  It could get noisy.

In SDRangel, the volume control is a round "knob" in the demodulator. Each demodulator has its own volume control.
The default squelch level is way too high in DSD Demod. Turn it down until you see noise in the little scope window, then turn it up to the point where the noise disappears. Use the cosine filter. You may need to adjust FMd to get it to work.
https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/demoddsd/readme.md
To be familiar with using SDRangel, use a broadcast FM demod and play with that.

Waymorehoss, 2024-09-02:
I will play around some more with sdrangel i did get audio from it earlier and i put it on a control channel i saw some decoding in the terminal but it was all skipping and stuff. i have played around with IFGR and RFGR in op25 but i still get this overload issue and it happens around every minute or so it's really messing up my listening all my radio's are doing it so it has to be with the antenna or something does your's do it to?

srs4511351, 2024-09-02:
SDRangel does not use control channels.
Control channels do not contain audio and are not listened to.
I think we are confused about this.

If the overload develops over time, it may be due to the hardware AGC in the SDR misbehaving. I do not have this problem.
There is a command to set the AGC threshold, but I will have to look it up when I get home.

Waymorehoss, 2024-09-02:
lol i don't know how i'm suppose to decode your email. you've lost me on that. 
i get that the gnuplot takes alot of cpu and probably dies out but i get this overload or interference on all software and on all computers. 
i have set my  IFGR to 15 and RFGR to 1 i am decoding a channel right now i can hear traffic i also have set it up with the web gui and that works to. here is a screenshots of just op25 running with gnuplot open and u can see how the gnuplot changes but this is across all my devices and hardware/software. oh and i know the control channel doesn't contain audio i  said that wrong, i have 3 dongles and the RSPdx it happens on all of them but maybe it is the SDR like u say.l
[image]
[image]

Waymorehoss, 2024-09-02:
Oh and i was gonna send video i can't add video here but i could on email

Waymorehoss, 2024-09-02:
i figured out you email i just emailed you.

boatbod, 2024-09-02:
I don't know SDR Angel's capabilities, but any sdr software that decodes p25 ph2 audio necessarily has to listen to the control channel in order to obtain required information to xor the scrambled voice frames.   Unlike p25 ph1, you can't simply tune into the voice channel and stream whatever you're receiving.

Waymorehoss, 2024-09-02:
To listen to NXDN and P25 in sdrangel do i just open the DSD demodulator ? and i dont' need to pipe the audio out do i?

srs4511351, 2024-09-02:
See: https://radiosrs.net/installing_SDRangel.html
In SDRangel
To add an SDR device, click the Add RX device icon on the left side, in the top bar just to the right of W0 (Workspace 0).
Select your SDR from the list and then click "OK" in the dialog

Near the top, in the Demodulator panel, click the icon just to the left of the SDR device name.
Select the DSD demodulator from the combo list and press the Apply button to add the channel to the passband. click the Close button.

Put t