boatbod/op25 #105: tcpdump instead of wireshark

frankmasio, 2021-02-18:
I am old fashion and use tcpdump  wireshark is so big i am able to see data with 

`tcpdump -i lo udp port 23456 -vv`

any tips to see anything intresting

frankmasio, 2021-02-18:
sounds like it does and sounds like its allot more work to code then wireshark

boatbod, 2021-02-20:
Long ago wireshark used to be the method of decoding the P25 FDMA protocol.  Gradually over time more in-depth protocol decoding was added to OP25 and eventually it became more capable than the old wireshark plugin.  These days, use of wireshark is deprecated and it actually takes considerable effort to build a plugin that works.  If and when you get it running, you'll find out that it does not support P25 TDMA signaling...

So anyway, to answer your question, wireshark *used* to show more than tcpdump, but either are capable of showing the raw undecoded dibit stream, not that it's of much interest.  Far better is to turn the op25 logging up to "-v 10" or maybe "-v 11" and see what pops out in the stderr.2 file.

closed 2021-02-20 by boatbod