boatbod/op25 #106: Audio clipping for certain users on system

W3AXL, 2021-02-22:
I've encountered an issue with certain subscribers on one of the systems I monitor. Some of the radios on the system have very loud audio compared to the majority. When these radios transmit, clipping occurs on my stream.

On previous versions of the boatbod fork, I was able to fix the issue by making the following change in rx.py:

```diff
- os.environ['IMBE'] = 'soft'
+ # os.environ['IMBE'] = 'soft'
```

However, after pulling the latest version, this change no longer works and users are again clipping.

A snippet of stream audio, with clipping, is here: https://soundcloud.com/user-486167826/op25-clipping

boatbod, 2021-02-25:
Using the legacy alternate codec implementation might work around your clipping problem but it causes other issues because there are things that only the default codec knows how to do.  e.g. AMBE tone generation.

I am curious to know whether you are using any PCM scaling (the -x command line option) as it is really unusual for the audio to clip with the default setting (-x 1.0).  Usually people complain that the audio is too quiet and as a result they artificially add gain to boost the volume, then when an occasional louder transmission comes though it does clip.

If you are able to capture a raw symbol dump of some clipped transmissions it would be useful for debugging purposes.  You can enable this by using the "-r filename.bin" command line option with rx.py.

W3AXL, 2021-02-25:
I am using the -x switch, however with a value of 0.5 to try and reduce the volume. It doesn't seem to have any effect on the clipping as far as I can tell.

I'm working on getting a symbol dump set up. The other challenge is getting it while a user with clipping audio is speaking.

What I don't understand is why commenting out the environment variable on my previous build of op25 (guessing a clone date of about a year ago or so) seemed to solve all the clipping issues and why it doesn't work now.

boatbod, 2021-02-25:
That "IMBE" environment variable is a hack that causes the op25 library code to utilize an alternative codec for phase 1 traffic.  Some time back, but within the last year, I made changes that significantly improved the efficiency of the phase 1 frame assembler and voice decoder by avoiding unnecessary bit shuffling conversions to/from bit arrays and byte arrays because it wasn't needed when working with the default codec.  The other codec has different interfaces which do require the (useless) bit shuffling, so enabling it is no longer an option.

W3AXL, 2021-03-04:
Was finally able to get a small snippet of clipped audio. Most of this symbol file is silence but at the end there was a short clipped transmission. Let me know if you need a longer sample than this.

(Had to rename to a .txt to let github upload it)
[symbols.txt](https://github.com/boatbod/op25/files/6084541/symbols.txt)

boatbod, 2021-03-05:
I have been able to replay the audio and do hear the brief click/disturbance which you refer to.  The dispatcher's audio is also noticeably "louder" than the mobile units which would be consistent with a clipping problem.  Question is whether the clipping is in the original transmission or op25's codec; I'm going to see if I can run the same sequence through the alternate codec to see if it plays any differently.

boatbod, 2021-03-07:
The issue is resolved on my local build.  Somewhat of a silly problem caused by one of the many redundant data manipulations which I am trying to eliminate for performance reasons.

closed 2021-03-07 by boatbod

W3AXL, 2021-03-08:
Haven't heard any clipping since I pulled down the latest version. Seems like that was the fix!

marklt1, 2021-03-08:
> Haven't heard any clipping since I pulled down the latest version. Seems like that was the fix!

I'm very happy with the results too. I had just assumed that the clipping issue was on the subscriber's end.

boatbod, 2021-03-09:
The fix was a somewhat accidental discovery; the floating point xMBE codec natively produces 16 bit samples in the range -32767 to +32767.  The old code had several redundant steps for dividing this number back into a float with range -1.0 to +1.0 to return to the calling function which summarily multiplied it back out to a 16 bit integer again.  Rather silly and somewhere along the way it introduced some rounding or sign errors which caused the clicks and pops at high amplitudes.

marklt1, 2021-03-09:
Great teamwork! Thanks to both of you!!!

W3AXL, 2021-07-28:
For future reference, the commit that addressed this fix is 27341e9b905417a4528551fb198d511fbf38697c