boatbod/op25 #108: CMakeLists makes wrong assumptions about python path

scresante, 2021-04-11:
Affected file: CMakeLists.txt 

I'm trying to put in work to make op25 compile and work across multiple distros, not just ubuntu. With that in mind, there is some code in the file that I feel can be fixed. 

This breaks compilation on archlinux: `m1 = os.path.join('lib', 'python' + sys.version[:3], 'dist-packages')`

The code assumes the directory for the python libraries is 'dist-packages'. 

I have simplified the code and tested it on ubuntu20.04 python2, and arch python3. It returns the correct results during the cmake build process. 

[0001-enable-distro-agnostic-python-library-for-part-of-cm.patch.txt](https://github.com/boatbod/op25/files/6291121/0001-enable-distro-agnostic-python-library-for-part-of-cm.patch.txt)

boatbod, 2021-04-27:
Thanks, I'll take a look and make sure it works on RPi

scresante, 2023-02-17:
I'm revisiting this process and using arch on arm (alarm) now. The same  problem cropped up. The solution is to use `cmake .. -DCMAKE_INSTALL_PREFIX="/usr"` in the buildchain.

closed 2023-02-17 by scresante