boatbod/op25 #110: trunk.tsv format deviations are not handled gracefully

scresante, 2021-04-12:
When rx.py parses trunk.tsv, some small mistakes in the file cause a crash when they don't need to, and the exception generated isn't too helpful. 

Spaces at the end of the line get passed to get_frequency, which expects a number, not blank spaces.

I have two simple fixes to address empty lines, and blank spaces at the ends of lines, which should be ignored.

I added the changes to my [extant pull request](https://github.com/boatbod/op25/pull/109t… sorry, I'm still re-learning git and how to do clean pull requests.

boatbod, 2021-04-27:
Thanks, I'll take a look.

As a general point I'd note that the rx.py version of op25 is deprecated and current development effort is being focused on multi_rx.py.  Configuration for that is .json format with the exception of the trunk-tags and blacklist/whitelist files.

boatbod, 2021-06-07:
I've reinforced the format checking and added the ability to include "#" comments

closed 2021-06-07 by boatbod