boatbod/op25 #112: /usr/include/gnuradio/block_gateway.h:20: Error: Unknown namespace 'pybind11'

MaxwellDPS, 2021-05-30:
Hey all,

Upon compiling against gnu radio 3.8 on oubuntu 20.04 I am getting the following error. Where should I start to try and fix this?

```
max@radio-ne-lnk-01:~/op25/build$ sudo make
[ 22%] Built target gnuradio-op25
[ 24%] Built target test-op25
[ 24%] Built target op25_swig_swig_doc
[ 25%] Swig source op25_swig.i
/usr/include/gnuradio/block_gateway.h:20: Error: Unknown namespace 'pybind11'
make[2]: *** [op25/gr-op25/swig/CMakeFiles/op25_swig_swig_compilation.dir/build.make:64: op25/gr-op25/swig/CMakeFiles/op25_swig.dir/op25_swigPYTHON.stamp] Error 1
make[2]: *** Deleting file 'op25/gr-op25/swig/CMakeFiles/op25_swig.dir/op25_swigPYTHON.stamp'
make[1]: *** [CMakeFiles/Makefile2:550: op25/gr-op25/swig/CMakeFiles/op25_swig_swig_compilation.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

boatbod, 2021-06-07:
Did you use the ~/op25/install.sh script after cloning the repo?
If not, my suggestion would be to nuke the ~/op25/build directory and then re-run install.sh

MaxwellDPS, 2021-06-09:
@boatbod Sorry for the delay, I did reclone the whole reop and run the install script, and Im still getting the pybind11 error

```
max@radio-ne-lnk-01:~/op25$ cat /etc/os-release
NAME="Ubuntu"
VERSION="20.04.2 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.2 LTS"
```

`Linux radio-ne-lnk-01 5.4.0-74-generic #83-Ubuntu SMP Sat May 8 02:35:39 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`

Error:
```
[ 25%] Swig source op25_swig.i
/usr/include/gnuradio/block_gateway.h:20: Error: Unknown namespace 'pybind11'
make[2]: *** [op25/gr-op25/swig/CMakeFiles/op25_swig_swig_compilation.dir/build.make:64: op25/gr-op25/swig/CMakeFiles/op25_swig.dir/op25_swigPYTHON.stamp] Error 1
make[2]: *** Deleting file 'op25/gr-op25/swig/CMakeFiles/op25_swig.dir/op25_swigPYTHON.stamp'
make[1]: *** [CMakeFiles/Makefile2:550: op25/gr-op25/swig/CMakeFiles/op25_swig_swig_compilation.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
[ 22%] Built target gnuradio-op25
[ 24%] Built target test-op25
[ 24%] Built target op25_swig_swig_doc
[ 25%] Swig source op25_swig.i
/usr/include/gnuradio/block_gateway.h:20: Error: Unknown namespace 'pybind11'
make[2]: *** [op25/gr-op25/swig/CMakeFiles/op25_swig_swig_compilation.dir/build.make:64: op25/gr-op25/swig/CMakeFiles/op25_swig.dir/op25_swigPYTHON.stamp] Error 1
make[2]: *** Deleting file 'op25/gr-op25/swig/CMakeFiles/op25_swig.dir/op25_swigPYTHON.stamp'
make[1]: *** [CMakeFiles/Makefile2:550: op25/gr-op25/swig/CMakeFiles/op25_swig_swig_compilation.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```
output
```
max@radio-ne-lnk-01:~$ rm -rf op25/
max@radio-ne-lnk-01:~$ git clone https://github.com/boatbod/op25.git
Cloning into 'op25'...
remote: Enumerating objects: 7888, done.
remote: Counting objects: 100% (406/406), done.
remote: Compressing objects: 100% (266/266), done.
remote: Total 7888 (delta 219), reused 281 (delta 136), pack-reused 7482
Receiving objects: 100% (7888/7888), 2.17 MiB | 10.39 MiB/s, done.
Resolving deltas: 100% (5260/5260), done.
max@radio-ne-lnk-01:~$ cd op25/
max@radio-ne-lnk-01:~/op25$ ./install.sh
Updating packages list
Hit:1 http://be.archive.ubuntu.com/ubuntu focal InRelease
Hit:2 https://download.docker.com/linux/ubuntu focal InRelease
Hit:3 http://us.archive.ubuntu.com/ubuntu focal InRelease
Hit:4 https://nvidia.github.io/libnvidia-container/stable/ubuntu18.04/amd64  InRelease
Hit:5 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  InRelease
Hit:6 https://nvidia.github.io/nvidia-docker/ubuntu18.04/amd64  InRelease
Get:7 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]
Hit:8 http://repo.feed.flightradar24.com flightradar24 InRelease
Hit:9 http://flightaware.com/adsb/piaware/files/packages buster InRelease
Hit:10 http://archive.ubuntu.com/ubuntu bionic InRelease
Hit:11 http://ppa.launchpad.net/gnuradio/gnuradio-releases/ubuntu focal InRelease
Get:12 http://us.archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]
Get:13 http://be.archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]
Hit:14 http://flightaware.com/adsb/piaware/files/packages jessie InRelease
Get:15 http://us.archive.ubuntu.com/ubuntu focal-backports InRelease [101 kB]
Get:16 http://be.archive.ubuntu.com/ubuntu focal-backports InRelease [101 kB]
Hit:17 http://ppa.launchpad.net/graphics-drivers/ppa/ubuntu focal InRelease
Get:18 http://be.archive.ubuntu.com/ubuntu focal-security InRelease [114 kB]
Get:19 https://esm.ubuntu.com/infra/ubuntu focal-infra-security InRelease [7,426 B]
Hit:20 https://opensky-network.org/repos/debian opensky InRelease
Get:21 https://esm.ubuntu.com/infra/ubuntu focal-infra-updates InRelease [7,425 B]
Hit:22 http://ppa.launchpad.net/myriadrf/drivers/ubuntu focal InRelease
Get:23 http://security.ubuntu.com/ubuntu focal-security/main amd64 DEP-11 Metadata [24.5 kB]
Get:24 http://security.ubuntu.com/ubuntu focal-security/universe amd64 DEP-11 Metadata [58.3 kB]
Get:25 http://security.ubuntu.com/ubuntu focal-security/multiverse amd64 DEP-11 Metadata [2,464 B]
[... 140 lines cut ...]
[ 22%] Built target gnuradio-op25
Scanning dependencies of target test-op25
[ 23%] Building CXX object op25/gr-op25/lib/CMakeFiles/test-op25.dir/test_op25.cc.o
[ 23%] Building CXX object op25/gr-op25/lib/CMakeFiles/test-op25.dir/qa_op25.cc.o
[ 24%] Linking CXX executable test-op25
[ 24%] Built target test-op25
Scanning dependencies of target op25_swig_swig_doc
[ 24%] Built target op25_swig_swig_doc
Scanning dependencies of target op25_swig_swig_compilation
[ 25%] Swig source op25_swig.i
/usr/include/gnuradio/block_gateway.h:20: Error: Unknown namespace 'pybind11'
make[2]: *** [op25/gr-op25/swig/CMakeFiles/op25_swig_swig_compilation.dir/build.make:64: op25/gr-op25/swig/CMakeFiles/op25_swig.dir/op25_swigPYTHON.stamp] Error 1
make[2]: *** Deleting file 'op25/gr-op25/swig/CMakeFiles/op25_swig.dir/op25_swigPYTHON.stamp'
make[1]: *** [CMakeFiles/Makefile2:550: op25/gr-op25/swig/CMakeFiles/op25_swig_swig_compilation.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
[ 22%] Built target gnuradio-op25
[ 24%] Built target test-op25
[ 24%] Built target op25_swig_swig_doc
[ 25%] Swig source op25_swig.i
/usr/include/gnuradio/block_gateway.h:20: Error: Unknown namespace 'pybind11'
make[2]: *** [op25/gr-op25/swig/CMakeFiles/op25_swig_swig_compilation.dir/build.make:64: op25/gr-op25/swig/CMakeFiles/op25_swig.dir/op25_swigPYTHON.stamp] Error 1
make[2]: *** Deleting file 'op25/gr-op25/swig/CMakeFiles/op25_swig.dir/op25_swigPYTHON.stamp'
make[1]: *** [CMakeFiles/Makefile2:550: op25/gr-op25/swig/CMakeFiles/op25_swig_swig_compilation.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

boatbod, 2021-06-09:
Do you have one of the "pybind11" packages installed?  Please run the following command and report the output.
    apt list --installed | grep pybind11

MaxwellDPS, 2021-06-09:
yep, looks like pybind11-dev
```
max@radio-ne-lnk-01:~/radio/NESR/legacy$ apt list --installed | grep pybind11

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

pybind11-dev/focal,focal,now 2.5.0-4~ubuntu20.04.1~ppa1 all [installed]
max@radio-ne-lnk-01:~/radio/NESR/legacy$
```

boatbod, 2021-06-10:
Unless you need that package I would uninstall it.
**sudo apt-get remove pybind11-dev
sudo apt-get autoremove**

MaxwellDPS, 2021-06-11:
@boatbod I did go ahead and remove pybind11-dev and run apt autoremove, it still failed at 25% with the same error

```
[ 25%] Swig source op25_swig.i
/usr/include/gnuradio/block_gateway.h:20: Error: Unknown namespace 'pybind11'
make[2]: *** [op25/gr-op25/swig/CMakeFiles/op25_swig_swig_compilation.dir/build.make:64: op25/gr-op25/swig/CMakeFiles/op25_swig.dir/op25_swigPYTHON.stamp] Error 1
make[2]: *** Deleting file 'op25/gr-op25/swig/CMakeFiles/op25_swig.dir/op25_swigPYTHON.stamp'
make[1]: *** [CMakeFiles/Makefile2:550: op25/gr-op25/swig/CMakeFiles/op25_swig_swig_compilation.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
[ 22%] Built target gnuradio-op25
[ 24%] Built target test-op25
[ 24%] Built target op25_swig_swig_doc
[ 25%] Swig source op25_swig.i
/usr/include/gnuradio/block_gateway.h:20: Error: Unknown namespace 'pybind11'
make[2]: *** [op25/gr-op25/swig/CMakeFiles/op25_swig_swig_compilation.dir/build.make:64: op25/gr-op25/swig/CMakeFiles/op25_swig.dir/op25_swigPYTHON.stamp] Error 1
make[2]: *** Deleting file 'op25/gr-op25/swig/CMakeFiles/op25_swig.dir/op25_swigPYTHON.stamp'
make[1]: *** [CMakeFiles/Makefile2:550: op25/gr-op25/swig/CMakeFiles/op25_swig_swig_compilation.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

boatbod, 2021-06-11:
It's not clear quite what the pybind11 package did to gnuradio, but when I look at the file "/usr/include/gnuradio/block_gateway.h" on my computer is has absolutely no mention of pybind11 anywhere within it.  From the error, I'm guessing yours is different at line 20.
```
/* -*- c++ -*- */
/*
 * Copyright 2011-2013,2017 Free Software Foundation, Inc.
 *
 * This file is part of GNU Radio
 *
 * GNU Radio is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation; either version 3, or (at your option)
 * any later version.
 *
 * GNU Radio is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with GNU Radio; see the file COPYING.  If not, write to
 * the Free Software Foundation, Inc., 51 Franklin Street,
 * Boston, MA 02110-1301, USA.
 */

#ifndef INCLUDED_RUNTIME_BLOCK_GATEWAY_H
#define INCLUDED_RUNTIME_BLOCK_GATEWAY_H

#include <gnuradio/api.h>
#include <gnuradio/block.h>
#include <gnuradio/feval.h>

namespace gr {

/*!
 * The work type enum tells the gateway what kind of block to
 * implement.  The choices are familiar gnuradio block overloads
 * (sync, decim, interp).
 */
enum block_gw_work_type {
    GR_BLOCK_GW_WORK_GENERAL,
    GR_BLOCK_GW_WORK_SYNC,
    GR_BLOCK_GW_WORK_DECIM,
    GR_BLOCK_GW_WORK_INTERP,
};

```

boatbod, 2023-01-23:
pybind11 supported for gnuradio-3.10 and above