boatbod/op25 #114: Thoughts on gnuradio 3.9?

brian-armstrong, 2021-07-19:
Hello,

First off, thank you for maintaining this fork!

Do you have plans for the migration to gnuradio 3.9? It looks like it's the default as of Ubuntu 21, and it does away with SWIG as you may be aware. https://wiki.gnuradio.org/index.php/GNU_Radio_3.9_OOT_Module_Porting_Guide

Cheers
Brian

boatbod, 2021-07-20:
Oh joy... more change for the sake of change!
Actually to be honest SWIG is a pain in the derriere so I won't be sad to say goodbye, but as yet I have no clue how much rework it will necessitate.

scresante, 2021-10-01:
How important would it be to have dual GR 3.8/3.9 compatibility? Would that be even worth it? I am playing around in the code now and cleaning up anything related to python2 or GR<3.8 as a start.

boatbod, 2021-10-01:
Backward compatibility has always been desirable since the speed of adoption of newer library versions (e.g. gnuradio) varies very widely across supported distros.

llamaonaskateboard, 2022-10-10:
I took @bkerler's fork from https://github.com/boatbod/op25/pull/149 and ported all of the other python SWIG modules to the new pybind style as a new fork over at http://github.com/llamaonaskateboard/op25

There's a few other modifications to get it to build on debian testing which currently has:
gnuradio 3.10.4.0-2+b1
gr-osmosdr 0.2.3-7+b1
cmake 3.24.2-1
python3 3.10.6-1

I also found once built that rx.py would hang and found a discussion and workaround at https://www.mail-archive.com/discuss-gnuradio@gnu.org/msg75225.html for this.
As a result, there's now local definitions and python bindings of message, msg_queue and msg_handler within gr::op25 with the add call_guard.

alphafox02, 2022-10-16:
I'd like to include this in a 22.04 based DragonOS I'm working on, has GR 3.10.4, gr-osmosdr etc. All working, some build from source. In 20.04 everything was working perfectly. Would it be safe to say I should be looking at your fork/repo @llamaonaskateboard as a test? I have some trunked systems I can try this with.

alphafox02, 2022-10-16:
@llamaonaskateboard haven't dug into it much yet, but noticed this after building/installing your fork on 3.10.4

`./rx.py -h
Using Python /usr/bin/python3
Traceback (most recent call last):
  File "/home/dragon/Downloads/op25/op25/gr-op25_repeater/apps/./rx.py", line 77, in <module>
    import gnuradio.op25 as op25
  File "/usr/local/lib/python3.10/dist-packages/gnuradio/op25/__init__.py", line 45, in <module>
    from .op25_python import *
ImportError: generic_type: type "decoder_bf" referenced unknown base type "gr::block"`

boatbod, 2022-10-16:
@llamaonaskateboard  I'm really short of free time to work on the port at present, but would be more than happy to open the repo to you so that you can contribute directly.

alphafox02, 2022-10-16:
@boatbod maybe have a maint-3.8 and maint-3.10 branch? By the way, thanks for the awesome project. If I can sort out my issues on 3.10 I’ll test it out some more.

lwvmobile, 2022-10-16:
If I may jump in here for a minute, I get nosy when I follow a project on Github and get a lot emails lol, but doesn't Maxx already have the Osmocom OP25 running on Python 3 and GNURadio 3.9 and 3.10. Went by the Osmocom branch a few days ago and saw some updates that would suggest as much, but haven't tested anything yet myself.

https://gitea.osmocom.org/op25/op25

> but would be more than happy to open the repo to you so that you can contribute directly.

Well, hey, while we are handing out repo keys, I'd love a chance to ~~deface~~ work on OP25

boatbod, 2022-10-16:
Yes, it's true that Osmocom op25 already supports gr-3.9, but osmocom and boatbod are actually very different in their capabilities.  Max and Outerdog went for a flashy gui and haven't really done as much with the feature capabilities, whereas I went for features and neglected the gui.  It's a long story, but the two codebases have diverged a long way since 2017 when they forked.

lwvmobile, 2022-10-16:
Ah okay, I wasn't quite sure how far the bases have diverged, or if the Osmocom could be used for clues on what to do if nothing else.

boatbod, 2022-10-16:
You can definitely get clues and concepts from osmo, but the interfaces have diverged significantly.  In all honesty I need to spend some time cleaning up/clearing out all the legacy code that's never going to be used.  e.g. transmit capability, wireshark support, logfile_workers etc, etc.  That's one of the reasons I never got around to the gr-3.9 port because in an ideal world the two would happen at the same time.

llamaonaskateboard, 2022-10-17:
> @llamaonaskateboard haven't dug into it much yet, but noticed this after building/installing your fork on 3.10.4
>
> `./rx.py -h Using Python /usr/bin/python3 Traceback (most recent call last): File [...]

@alphafox02 That's odd, the gr::block python binding comes from gnuradio (https://github.com/gnuradio/gnuradio/blob/maint-3.10/gnuradio-runtime/python/gnuradio/gr/bindings/block_python.cc).
On debian, it's provided by the main gnuradio package:
```
$ dpkg -S gr_python.cpython-310-x86_64-linux-gnu.so
gnuradio: /usr/lib/python3/dist-packages/gnuradio/gr/gr_python.cpython-310-x86_64-linux-gnu.so
$ dpkg -l gnuradio*
||/ Name                              Version            Architecture Description
+++-=================================-==================-============-======================================================================
ii  gnuradio                          3.10.4.0-2+b1      amd64        GNU Radio Software Radio Toolkit
ii  gnuradio-dev                      3.10.4.0-2+b1      amd64        GNU Software Defined Radio toolkit development
```
What happens if you just do `python3 -c "from gnuradio import gr"`?

llamaonaskateboard, 2022-10-18:
> @llamaonaskateboard I'm really short of free time to work on the port at present, but would be more than happy to open the repo to you so that you can contribute directly.

@boatbod I wish I could help out but also short on time. 
I think a key issue is going to be how to deal with backwards compatibility with earlier gnuradio versions since SWIG to pybind is quite a change. cmake also has some changes between versions, particularly with path variables.
osmocom currently handles this by keeping swig and having a different build script for 3.9+ that copies around parts of the source tree, modifies cmake files and generates the python bindings live (https://gitea.osmocom.org/op25/op25/src/branch/master/add_gr3.9.py), which seems ugly but works I guess.

Side note, I've since found https://github.com/robotastic/trunk-recorder/ which actually fits my current use case a bit more. 
It also supports gnuradio 3.7-3.10 so might be able to get some backwards compatibility ideas from it.

boatbod, 2022-10-18:
Understood, and thanks for your work so far.
My approach to cmake has always been to make the build script intelligent enough to make it's own decisions rather than expect the end user to take on that task.  This would be no different, if indeed it is even possible to auto-adapt such a large change in interfaces.

llamaonaskateboard, 2022-10-18:
@alphafox02 I just built using that ISO you linked and it...just worked...
My rx.py command line is `./rx.py --nocrypt --args "rtl" --gains 'lna:36' -S 960000 -X -q 0 -v 1 -2 -V -U -T trunk.tsv -q 72.62 -v 10 -l http:0.0.0.0:8080` for reference.

I just committed some minor tweaks (pydoc/doxygen stuff) to get it to build on regular Ubuntu 22.04 and resynced with boatbod's repo so maybe give it a go now.

alphafox02, 2022-10-18:
Perfect! I’ll give it a try and add it all there by default. Thank you for taking a look at the ISO. I’m trying to test each thing little by little.

alphafox02, 2022-10-18:
Errors are gone and I suspect, like you, I’ll be able to run it now once I get back and plug in an SDR. Thanks!

boatbod, 2023-01-02:
I've added a new experimental "gr310" branch to my repo.  It's basically the @llamaonaskateboard fork with a tweak to the install.sh script so that it builds for gnuradio-3.10 (only).  Right now it does run and decode voice, but I'm seeing core dumps at shutdown and also one or two unexplained crashed mid-call.

boatbod, 2023-01-09:
Found a nasty little bug in gnuradio's fll_band_edge_cc filter that was causing my spurious core dumps on gr310 branch.  Basically it seems the gnuradio team forgot to add mutex locks to make the block thread safe, so for the time being I've implemented a work-around in p25_demodulator.py

Still doing some cleanup on the gr310 branch before it's going to be worth trying.

boatbod, 2023-01-13:
Branch "gr310" is now working on Ubuntu 22.04 with gnuradio-3.10

Closing this issue, with much appreciation for all the help getting the port started.

closed 2023-01-13 by boatbod

alphafox02, 2023-01-13:
Thanks for creating the branch! I’ll add it to the list.