boatbod/op25 #115: Web interface encryption options

djkraven, 2021-07-25:
It is possible to configure the web interface to HTTPS connection instead of HTTP?

boatbod, 2021-08-14:
Not as far as I know.

closed 2021-08-14 by boatbod

djkraven, 2021-08-14:
I stood up a secure reverse apache proxy and that works for everything 
except the audio which is different issue. I would like to stream to 
broadcastify, direct, and local speakers. Currently the local speakers and 
broadcastify are working but they don't seem to be a one for one.

boatbod, 2021-08-14:
You could also set up an ssh tunnel for the web browser port.
Not sure what you mean by local speakers and broadcastify not being one for one?  If you are using liquidsoap, the local audio will be the same as that sent to broadcastify, but without the delay that icecast seems to add to everything,

djkraven, 2021-08-14:
It seems I hear completely different things from the local and 
broadcastify even taking a delay into account. The ssh tunnel is not 
going to address the https certificate method. the Apache (nginx or 
other http proxy solutions) resolves the issue and allow me to setup a 
page of iframes with youtube feeds. A mix of http and https pages don't 
with properly they all need to be https.

Thank you for the replies.