boatbod/op25 #117: Can't get smartnet system to work

M4ttjabz, 2021-08-28:
I've been able to successfully set-up my local P16 Smartnet System in Windows with both Unitrunker V2 and DSD Fastlane but I'd really like to use it with my linux box.  I've tried several attempts at re-installing and keep getting the same error.

Here's my json file:
{
    "channels": [
        {
            "name": "control channel", 
            "trunking_sysname": "Example",
            "device": "rtl0",
            "raw_output": "",
            "demod_type": "fsk", 
            "destination": "smartnet", 
            "excess_bw": 0.35, 
            "filter_type": "fsk2", 
            "if_rate": 18000, 
            "plot": "mixer",
            "enable_analog": "off",
            "symbol_rate": 3600
        },
        {
            "name": "voice channel", 
            "trunking_sysname": "Example",
            "meta_stream_name": "stream_0",
            "device": "rtl1",
            "blacklist": "",
            "whitelist": "",
            "raw_output": "",
            "demod_type": "fsk4", 
            "destination": "udp://127.0.0.1:23456", 
            "enable_analog": "auto",
            "nbfm_deviation": 4000,
            "nbfm_squelch_threshold": -60,
            "nbfm_squelch_gain": 0.0050,
            "nbfm_raw_output": "",
            "nbfm_enable_subchannel": false,
            "excess_bw": 0.2, 
            "filter_type": "widepulse", 
            "if_rate": 24000, 
            "plot": "",
            "symbol_rate": 4800
        }
    ], 
    "devices": [
        {
            "args": "rtl=0", 
            "frequency": 8572625000, 
            "gains": "LNA:39", 
            "name": "rtl0", 
            "offset": 0, 
            "ppm": 0.0, 
            "rate": 1000000, 
            "tunable": true
        },
        {
            "args": "rtl=1", 
            "frequency": 857262500, 
            "gains": "LNA:39", 
            "name": "rtl1", 
            "offset": 0, 
            "ppm": 0.0, 
            "rate": 1000000, 
            "tunable": true
        }
    ],
    "trunking": {
        "module": "tk_smartnet.py",
        "chans": [
            {
                "sysname": "Example",
                "control_channel_list": "857.2625",
                "tgid_tags_file": "",
                "tgid_hold_time": 2.0,
                "blacklist": "",
                "whitelist": "",
                "bandplan_comment": "Set bandplan to one of '400' '800' '800_reband' '900'",
                "bandplan": "800_reband",
            }
        ]
    },
    "metadata": {
        "module": "icemeta.py",
        "streams": [
            {
                "stream_name": "stream_0",
                "meta_format_idle": "[idle]",
                "meta_format_tgid": "[%TGID%]",
                "meta_format_tag":  "[%TGID%] %TAG%",
                "icecastServerAddress": "your_stream_host_and_port",
                "icecastMountpoint": "your_mountpoint_name",
                "icecastPass": "your_password",
                "icecastMountExt": ".xspf",
                "delay": 0.0
            },
            {
                "stream_name": "stream_1",
                "meta_format_idle": "[idle]",
                "meta_format_tgid": "[%TGID%]",
                "meta_format_tag":  "[%TGID%] %TAG%",
                "icecastServerAddress": "your_second stream_host_and_port",
                "icecastMountpoint": "your_second_mountpoint_name",
                "icecastPass": "your_second_password",
                "icecastMountExt": ".xspf",
                "delay": 0.0
            }
        ]
    },
    "audio": {
        "module": "sockaudio.py",
        "instances": [
            {
                "instance_name": "audio0",
                "device_name": "default",
                "udp_port": 23456,
                "audio_gain": 1.0,
                "number_channels": 1
            }
        ]
    },
    "terminal": {
        "module": "terminal.py",
        "terminal_type": "curses",
        "#terminal_type": "http:127.0.0.1:8080",
        "curses_plot_interval": 0.2,
        "http_plot_interval": 1.0,
        "http_plot_directory": "../www/images",
        "tuning_step_large": 1200,
        "tuning_step_small": 100
    }
}

And the error I get is:

Using Python /usr/bin/python3
Traceback (most recent call last):
  File "./multi_rx.py", line 967, in <module>
    rx = rx_main()
  File "./multi_rx.py", line 931, in __init__
    config = json.loads(open(options.config_file).read())
  File "/usr/lib/python3.8/json/__init__.py", line 357, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.8/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.8/json/decoder.py", line 353, in raw_decode
    obj, end = self.scan_once(s, idx)
json.decoder.JSONDecodeError: Expecting property name enclosed in double quotes: line 74 column 13 (char 2216)

I'm using a Neoelec SDR and a HackRF and the control channel comes in perfectly; with no ppm needed. 

Any ideas?  I'm using the latest DragonOS. 

Thanks!

-Matt

boatbod, 2021-08-29:
Your cfg.json is malformed. Try taking out the trailing comma on line 74.  It's on the line containing "800_reband"

M4ttjabz, 2021-08-30:
Thanks man!!! I tried just deleting the , and it still didn't work; I guess it didn't like the HackRF.  Once I swapped out for another regular RTL Standard dongle; it works like a charm.  Any idea what identifier I should use for the HackRF? , it's clearly not Rtl=01.

I really appreciate all the work you put into this and the fact you respond to questions by linux-newbies like me.

boatbod, 2021-08-30:
I've not personally used a HackRF but perusing the gr-osmosdr makes me think you should substitiute "rtl=0" with "hackrf".  The "gains" parameter(s) may be different too.  Looks like the three arg values are RF, IF, and BB. Not sure what numeric range is appropriate.  Don't forget to also set your sample rate to something the HackRF will be happy with.

closed 2021-08-30 by boatbod