boatbod/op25 #118: No audio getting piped to alsa or pulseaudio

gettyhub, 2021-08-30:
Had pavucontrol open and noticed even though python was opening an audio stream, nothing was in the audio stream.  Not pushing anything through pulse or alsa.  Then I updated RPi 4 to and now there isn’t even an audio stream in pavucontrol.  Doesn’t work with any output audio jack or HDMI either one.  Frequencies and talk groups show up, but no audio.  Using just the setfrequency file to set the frequency.  The trunk.csv is just a tag file and is not needed to hear the audio (I hope).

gettyhub, 2021-08-30:
Even though autoppm was being used, maybe manually setting ppm is still better.  

I also wiped out the old installation, and redid a git clone fresh and audio is working.

boatbod, 2021-08-30:
op25 can communicate with the sound subsystem through either the "pulse" or "alsa" libraries.  For reasons best known to linux, sometimes one works better than the other, but seemingly without any sane reason.  Best to try both...

trunk-tags.tsv is not required unless you want tgid priority.
trunk.tsv is mandatory if you want trunking to function

closed 2021-08-30 by boatbod