boatbod/op25 #119: Talkgroup name display too brief

gitn8zpj, 2021-08-31:
Hi,
I think this would be useful for all...
The name display on the web interface and the CL interface displays the Talkgroup name only when the TG is actively transmitting.  A lot of times this is for only a second.  By the time I look at the display, it is gone, and I don't know who was talking.

Is there a way to have a scrolling "last heard" TG name display?  Or the last 5 heard?

Thank you!
- Scott -

boatbod, 2021-09-09:
Probably something best implemented in the new gui being developed

NebraskaCoder, 2021-09-12:
> Hi,
> I think this would be useful for all...
> The name display on the web interface and the CL interface displays the Talkgroup name only when the TG is actively transmitting. A lot of times this is [...]

How about a history section that shows a history of the last x amount of calls (TG ID, Radio RD, System, etc...)? I say x amount of calls so you can adjust the value based on how much data you want to store in memory.

gitn8zpj, 2021-09-12:
Gabriel,

Thank you for responding.

That sounds great !

Thank you,
  - Scott