boatbod/op25 #120: OP25 TSBKS line stops counting. Does not Return to Control Channel

simplio, 2021-09-07:
Hi

I am running OP25 Boatb0d branch on Lubuntu 20.4.3 LTS on Virtual Box. Host is Windows 10 machine.

I watch the tsbks stop counting then no more trunk following. It's a APCO P25 PII - TDMA system built by L3 Harris. https://www.radioreference.com/apps/db/?sid=10198

stderr.2 output is

Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "./rx.py", line 974, in run
    self.callback(msg)
  File "/home/sunplio/op25/op25/gr-op25_repeater/apps/trunking.py", line 1637, in process_qmsg
    updated += self.trunked_systems[nac].decode_tdma_msg(s, curr_time)
  File "/home/sunplio/op25/op25/gr-op25_repeater/apps/trunking.py", line 859, in decode_tdma_msg
    self.update_voice_frequency(f1, tgid=ga1, tdma_slot=self.get_tdma_slot(ch1), srcaddr=sa1)
NameError: name 'sa1' is not defined
09/07/21 10:39:39.942794 clear tgid=419, freq=853.600000, slot=0
09/07/21 10:39:39.942965 clear tgid=292, freq=853.600000, slot=1
Flowgraph completed. Exiting

Python 3 is installed by the install script

It will decode for 10 seconds then the error occurs.sometimes it takes longer other times shorter. This looks like a python 3 error to me because I searched " NameError " and i keep finding Python 3 errors related to it on various websites. I am just an appliance operator not knowing any of it but really want this to work.

Good day and thanks for listening.

boatbod, 2021-09-08:
Oops, I see a bug in there. Let me take a look.

simplio, 2021-09-08:
thanks for seeing that! I am looking forward to rebuilding op25 once you have the fix.

boatbod, 2021-09-08:
Fix pushed.  It's python only, so just "git pull" and restart the app. No need to recompile.

simplio, 2021-09-09:
> Fix pushed. It's python only, so just "git pull" and restart the app. No need to recompile.

TSBKS has surpassed 10000  still counting,  now returns to the control channel. Appears to be running smoothly on the Lubuntu LXQt virtual machine in Virtual Box. I am also amazed how much this i7 10th gen laptop can handle. Thanks boatbod the fix worked! 

If anymore unforeseen circumstance(s) with the software arises I will report it here.

Good day and many best wishes!
simplio

closed 2021-09-09 by simplio