boatbod/op25 #127: CMake error after install?

turdferguson68, 2021-09-30:
I'm running ubuntu 20.04. I installed boatbod/op25 about a week ago and had it running for about an hour this morning when suddenly it stopped working and said something like it couldnt pull gru to gnuradio. I updated gnuradio with their offical repository and when I went to run ./rebuild.sh got the following error:
-- Configuring incomplete, errors occurred!
See also "/home/spooby/op25/build/CMakeFiles/CMakeOutput.log".
See also "/home/spooby/op25/build/CMakeFiles/CMakeError.log".
make: *** No targets specified and no makefile found. Stop.
make: *** No rule to make target 'install'. Stop.

I've attached that full output and those .logs
I'm currently updating cmake to 3.21.3; I will keep you apprised

[cmakebuilderror.txt](https://github.com/boatbod/op25/files/7261990/cmakebuilderror.txt)
[CMakeOutput.txt](https://github.com/boatbod/op25/files/7261991/CMakeOutput.txt)
[CMakeError.txt](https://github.com/boatbod/op25/files/7261992/CMakeError.txt)

turdferguson68, 2021-09-30:
also, what is the process to follow if I want to uninstall and start from scratch or install a different op25?

turdferguson68, 2021-09-30:
No joy updating cmake. boatbod, you replied to my identical post on radioreference but my posts are in purgatory since I have a low post count.
new logs attached
[cmakebuilderror2.txt](https://github.com/boatbod/op25/files/7262088/cmakebuilderror2.txt)
[CMakeOutput2.txt](https://github.com/boatbod/op25/files/7262089/CMakeOutput2.txt)
[CMakeError2.txt](https://github.com/boatbod/op25/files/7262090/CMakeError2.txt)

boatbod, 2021-10-01:
It looks like you have an incompatible version of gnuradio.
```
-- Checking for module 'gnuradio-runtime'
--   Found gnuradio-runtime, version 3.10.0git
```
Version 3.9 or later drops support for SWIG which is currently the method used by op25 to interface between python and c++

closed 2021-10-01 by boatbod