boatbod/op25 #128: Compilation fails on archlinux

scresante, 2021-10-01:
Compilation on archlinux fails as of 09-30-21. 

I'm not asking for help, really, but I am documenting the problem here as I find a solution to it, perhaps followed by a pull request if I'm successful. 

[cmake.log](https://github.com/boatbod/op25/files/7263835/cmake.log)
[CMakeError.log](https://github.com/boatbod/op25/files/7263837/CMakeError.log)
[CMakeOutput.log](https://github.com/boatbod/op25/files/7263838/CMakeOutput.log)
[PKGBUILD.txt](https://github.com/boatbod/op25/files/7263840/PKGBUILD.txt)

These are my dependency versions:
```
gnuradio 3.9.2.0-3
gnuradio-osmosdr 0.2.3-4
boost 1.76.0-1
libpcap 1.10.1-1
gnuradio-osmosdr 0.2.3-4
rtl-sdr 1:0.8.0-3
swig 4.0.2-2
boost 1.76.0-1
python-numpy 1.21.2-1
python-waitress 2.0.0-1
python-requests 2.26.0-1
gnuplot 5.4.2-1
itpp 4.3.1-1
cmake 3.21.3-1
git 2.33.0-1
cppunit 1.15.1-2
pkgconf 1.8.0-1
doxygen 1.9.2-1
```

boatbod, 2021-10-01:
gnuradio-3.9 drops swig support, so it is fundamentally incompatible with op25 as currently written.

FwappyPotato, 2022-01-17:
also having this issue

boatbod, 2022-01-17:
As noted on 10/1/2021, op25 will need to be modified to be used with gnuradio-3.9, so until then it is not going to be compatible with archlinux unless you can drop back to an earlier version of GR.

scresante, 2022-03-03:
This can be closed, since it is a general issue addressed by #134  et al

boatbod, 2023-02-12:
OP25 can be run on arch, but the installer is not (yet) automated.  You'll need to install the following packages to build the gr310 branch.
```
sudo packman -Sy gnuradio gnuradio-osmosdr rtl-sdr libuhd git cmake cppunit doxygen boost libpcap orc base-devel clang pkgconf pybind11 python-numpy python-waitress python-setuptools gnuplot libsndfile spdlog hackrf
yay -S itpp castxml python-pygccxml
```

closed 2023-02-12 by boatbod

scresante, 2023-02-16:
FYI, just for your future archlinux safety, you don't ever want to use `pacman -Sy package` to install a package [partial upgrades unsupported](https://wiki.archlinux.org/title/System_maintenance#Partial_upgrades_are_unsupported)

Arch has its own challenges being a such a rapidly moving distribution. As of the time of this writing, boost-libs has already moved to 1.81, breaking gnuradio. But that's nothing to do with you. 

Thanks for the tip!

scresante, 2023-02-17:
One further amendment to the build instructions above: 

doing `cmake .. -DCMAKE_INSTALL_PREFIX="/usr"` will solve some problems with the default alarm install not getting paths right.