boatbod/op25 #141: multi_rx Airspy R2 P25 lockup

carbinefreak, 2021-12-08:
I'm attempting to run multi_rx on a 6 channel site and listen to all channels simultaneously. The control channel decodes and I get the first voice channel but after 30s the shell starts printing out "0"s since it can't get any data from the SDR. After unplugging and re-plugging the Airspy it comes back for another ~30s before crashing. Eventually the Kernel can't talk to the airspy any more and a host reboot is the only fix. 

https://www.radioreference.com/apps/db/?siteId=8177

`{
    "channels": [
        {
            "name": "Voice_ch1",
            "device": "sdr0",
            "trunking_sysname": "DTRS",
            "meta_stream_name": "stream_1",
            "demod_type": "cqpsk", 
            "cqpsk_tracking": true,
            "tracking_threshold": 30,
            "tracking_feedback": 0.85,
            "destination": "udp://127.0.0.1:23100",
            "excess_bw": 0.2,
            "filter_type": "rc",
            "frequency": 851300000,
            "if_rate": 24000,
            "plot": "symbol",
            "symbol_rate": 4800,
            "enable_analog": "off",
            "blacklist": "",
            "whitelist": ""
        },
        {
            "name": "Voice_ch2",
            "device": "sdr0",
            "trunking_sysname": "DTRS",
            "meta_stream_name": "stream_2",
            "demod_type": "cqpsk", 
            "cqpsk_tracking": true,
            "tracking_threshold": 30,
            "tracking_feedback": 0.85,
            "destination": "udp://127.0.0.1:23200",
            "excess_bw": 0.2,
            "filter_type": "rc",
            "frequency": 851825000,
            "if_rate": 24000,
            "plot": "constellation",
            "symbol_rate": 4800,
            "enable_analog": "off",
            "blacklist": "",
            "whitelist": ""
        },
        {
            "name": "Voice_ch3",
            "device": "sdr0",
            "trunking_sysname": "DTRS",
            "meta_stream_name": "stream_3",
            "demod_type": "cqpsk",
            "cqpsk_tracking": true,
            "tracking_threshold": 30,
            "tracking_feedback": 0.85,
            "destination": "udp://127.0.0.1:23300",
            "excess_bw": 0.2,
            "filter_type": "rc",
            "frequency": 852387500,
            "if_rate": 24000,
            "plot": "constellation",
            "symbol_rate": 4800,
            "enable_analog": "off",
            "blacklist": "",
            "whitelist": ""
        },
        {
            "name": "Voice_ch5",
            "device": "sdr0",
            "trunking_sysname": "DTRS",
            "meta_stream_name": "stream_5",
            "demod_type": "cqpsk",
            "cqpsk_tracking": true,
            "tracking_threshold": 30,
            "tracking_feedback": 0.85,
            "destination": "udp://127.0.0.1:23500",
            "excess_bw": 0.2,
            "filter_type": "rc",
            "frequency": 853525000,
            "if_rate": 24000,
            "plot": "constellation",
            "symbol_rate": 4800,
            "enable_analog": "off",
            "blacklist": "",
            "whitelist": ""
        },
        {
            "name": "Voice_ch6",
            "device": "sdr0",
            "trunking_sysname": "DTRS",
            "meta_stream_name": "stream_6",
            "demod_type": "cqpsk",
            "cqpsk_tracking": true,
            "tracking_threshold": 30,
            "tracking_feedback": 0.85,
            "destination": "udp://127.0.0.1:23600",
            "excess_bw": 0.2,
            "filter_type": "rc",
            "frequency": 856687500,
            "if_rate": 24000,
            "plot": "constellation",
            "symbol_rate": 4800,
            "enable_analog": "off",
            "blacklist": "",
            "whitelist": ""
        }
    ], 
    "devices": [
        {
            "args": "airspy=0",
            "frequency": 853000000,
            "gains": "LNA:12,MIX:12,IF:12",
            "gain_mode": false,
            "name": "sdr0",
            "offset": 0,
            "ppm": 0.03,
            "rate": 10000000,
            "usable_bw_pct": 0.95,
            "tunable": false
        }
    ],
    "trunking": {
        "module": "tk_p25.py",
        "chans": [
            {
                "nac": "0x0",
                "sysname": "DTRS",
                "control_channel_list": "852.925",
                "whitelist": "",
                "blacklist": "",
                "tgid_tags_file": "dtrs_tgs.tsv",
                "rid_tags_file": "",
                "tdma_cc": false,
                "crypt_behavior": 0
            }
        ]
    },
    "terminal": {
        "module": "terminal.py",
        "terminal_type": "curses",
        "curses_plot_interval": 0.1,
        "http_plot_interval": 1.0,
        "http_plot_directory": "../www/images",
        "tuning_step_large": 1200,
        "tuning_step_small": 100
    }
}
`

boatbod, 2021-12-19:
The "O" symbols in the console window are an indication of GNU Radio buffer overrun.  It's typically caused when the CPU cannot keep up with the processing demands of multiple decoders running simultaneously.  First of all eliminate all the plots to save cpu cycles, but even so you are going to need a powerful machine to run 6 channels simultaneously with a 10Mhz sample rate.

closed 2021-12-19 by boatbod

carbinefreak, 2021-12-19:
Thanks, this is running on a HP z620 with 24 cores and 64GB of ram so I would be impressed it would need more. I'll do some troubleshooting and see if i'm pegging a core while this runs.

boatbod, 2021-12-19:
Something is getting overloaded and causing samples not to move through the flowgraph in a timely manner.  Those "O" messages are produced by the gnuradio runtime when samples are lost due to oveeruns.  Most of the real time code in both gnuradio and op25 is written in c/c++, but the higher levels (call control) messages are passed up to the Python trunking module.  I'd be surprised if Python is directly causing the bottleneck though; the higher cpu utilization threads are usually associated with filtering and voice decoding, both of which are very floating-point intensive operations.