boatbod/op25 #147: Plots not working

kbbj0197, 2022-02-23:
Not an issue persay, but a query: what libraries are required to operate plots? op25 worked well under Ubuntu-16, but now using more restricted Linux version without multiple unneeded files/libraries. Currently op25 starts up, but will not call up any plots when #s are selected.

kbbj0197, 2022-02-26:
I received help on RadioReference.com:

The problem was that the app makes a call for the "gnuplot" file, but I did not have that file link. It was only listed as gnuplot4-xll with the links gnuplot and gnuplot4 linked to nothing. I relinked these to the gnuplot4-xll file and all works as advertised again. So I had all the libraries, but not properly referenced.

Thank you.

boatbod, 2022-05-10:
op25 assumes there is an executable called '/usr/bin/gnuplot'.  The minimum package install is gnuplot-x11

closed 2022-05-10 by boatbod